ni/nimi-python #106: Some public enum values start with _

marcoskirsch, 2017-07-28:
NI-DMM has a few enum values that start with a number: i.e. NIDMM_ATTR_ID_2_WIRE_RESISTANCE
Currently in Python this is nidmm.Function._2_WIRE_RESISTANCE
But leading _ is reserved for private which is not the case here. We should look for alternatives.

Other MI drivers have similar problems:

NIFGEN_ATTR_5450_OSCILLATOR_PHASE_DAC_VALUE
NISCOPE_ATTR_5102_ADJUST_PRETRIGGER_SAMPLES
NISCOPE_ATTR_5V_OUT_OUTPUT_TERMINAL
NISCOPE_ATTR_5102_ADJUST_PRETRIGGER_SAMPLES
...

DAQmx API spells out the numbers. This would work for NIDMM_ATTR_ID_2_WIRE_RESISTANCE but be really awkward for things such as NIFGEN_ATTR_5450_OSCILLATOR_PHASE_DAC_VALUE.

Fladolcetta, 2017-08-31:
Is this an issue with the Meta-data or is there another issue? Couldn't we just change the meta data in enum.py to remove the underscore? 
This would change "nidmm.Function._2_WIRE_RESISTANCE" to "nidmm.Function.2_WIRE_RESISTANCE"

marcoskirsch, 2017-10-17:
I propose we modify the generator to spit out a warning whenever it encounters these. That will at least allow us to compile a full list of the offending attribute names. Once we have that, we can think of a good solution.

marcoskirsch, 2017-10-19:
From [documentation](https://docs.python.org/3.6/library/enum.html?highlight=enum#allowed-members-and-attributes-of-enumerations)

> The rules for what is allowed are as follows: names that start and end with a single underscore are reserved by enum and cannot be used; all other attributes defined within an enumeration will become members of this enumeration, with the exception of special methods (__str__(), __add__(), etc.) and descriptors (methods are also descriptors).

Fladolcetta, 2017-11-06:
Lets see if nidaqmx-python and nixnet-python ran into this, and see how they dealt with it.

epage, 2017-11-09:
Yes, in nidaqmx.  I can't remember for nixnet.  They mirrored our C# API which had the same problem  In most cases, that meant using a word ("FOUR" instead of "4")  I don't think we add leading model numbers which complicates things

As for alternative solutions, I strongly recommend against a leading `_`.  A leading `_` indicates it is private / protected / implementation detail and anyone using it might break

marcoskirsch, 2017-11-09:
> As for alternative solutions, I strongly recommend against a leading _. A leading _ indicates it is private / protected / implementation detail and anyone using it might break

That's precisely what prompted this issue! Thanks for confirming we should indeed handle it.

marcoskirsch, 2017-11-09:
I am leaning towards the sad realization that we will have to source overrides for the enum value names in the Python metadata. This was recently enabled by https://github.com/ni/nimi-python/pull/575

The problem, however, is that the metadata under src/<driver>/metadata/enums.py already has prefixes stripped off and already leaves an underscore at the beginning of some of them. I think this policy is in the wrong place: nimi-python should be making that decision, not the extraction tools internal to NI which should have minimum policy.

So here's my proposal:

1. Change extraction tools to stop stripping off enum value prefix.
2. Commit new enum metadata
3. Modify nimi-python to remove the driver specific prefix (something like NIFGEN_VAL_). This will be specified in config.py metadata for the driver, or can be assumed by looking for the common prefix in _all_ the enum values for _all_ the enums.
4. either:
    4a. Let the interpreter eventually error when it runs into these or
    4b. Have checks and let the generator log warnings for all of these (more work)
5. Manually enter name overrides in enums_addon.py, for example only:
    * 5450_OSCILLATOR_PHASE_DAC_VALUE ->OSCILLATOR_PHASE_DAC_VALUE_5450
    * 5102_ADJUST_PRETRIGGER_SAMPLES  -> ADJUST_PRETRIGGER_SAMPLES_5102
    * 5V_OUT_OUTPUT_TERMINAL  -> OUTPUT_TERMINAL_5V_OUT
    * 5102_ADJUST_PRETRIGGER_SAMPLES -> ADJUST_PRETRIGGER_SAMPLES_5102

This is unfortunate but I think it's the way to go.

marcoskirsch, 2017-11-09:
The list of problematic enums as it currently starts in master:

```
Kirsch0:nimi-python kirsch$ find * | grep metadata/enums\.py | xargs grep \'name\'\:\ \'_
src/niscope/metadata/enums.py:                'name': '_48_TAP_STANDARD',
src/niscope/metadata/enums.py:                'name': '_48_TAP_HANNING',
src/niscope/metadata/enums.py:                'name': '_16_TAP_HANNING',
src/niscope/metadata/enums.py:                'name': '_8_TAP_HANNING',
src/niscope/metadata/enums.py:                'name': '_8_BITS',
src/niscope/metadata/enums.py:                'name': '_16_BITS',
src/niscope/metadata/enums.py:                'name': '_24_BITS',
src/niscope/metadata/enums.py:                'name': '_32_BITS',
src/niscope/metadata/enums.py:                'name': '_480I_59_94_FIELDS_PER_SECOND',
src/niscope/metadata/enums.py:                'name': '_480I_60_FIELDS_PER_SECOND',
src/niscope/metadata/enums.py:                'name': '_480P_59_94_FRAMES_PER_SECOND',
src/niscope/metadata/enums.py:                'name': '_480P_60_FRAMES_PER_SECOND',
src/niscope/metadata/enums.py:                'name': '_576I_50_FIELDS_PER_SECOND',
src/niscope/metadata/enums.py:                'name': '_576P_50_FRAMES_PER_SECOND',
src/niscope/metadata/enums.py:                'name': '_720P_50_FRAMES_PER_SECOND',
src/niscope/metadata/enums.py:                'name': '_720P_59_94_FRAMES_PER_SECOND',
src/niscope/metadata/enums.py:                'name': '_720P_60_FRAMES_PER_SECOND',
src/niscope/metadata/enums.py:                'name': '_1080I_50_FIELDS_PER_SECOND',
src/niscope/metadata/enums.py:                'name': '_1080I_59_94_FIELDS_PER_SECOND',
src/niscope/metadata/enums.py:                'name': '_1080I_60_FIELDS_PER_SECOND',
src/niscope/metadata/enums.py:                'name': '_1080P_24_FRAMES_PER_SECOND',
src/nidcpower/metadata/enums.py:                'name': '_50',
src/nidcpower/metadata/enums.py:                'name': '_60',
src/nidmm/metadata/enums.py:                'name': '_1_MICROAMP',
src/nidmm/metadata/enums.py:                'name': '_10_MICROAMP',
src/nidmm/metadata/enums.py:                'name': '_100_MICROAMP',
src/nidmm/metadata/enums.py:                'name': '_1_MILLIAMP',
src/nidmm/metadata/enums.py:                'name': '_3_5',
src/nidmm/metadata/enums.py:                'name': '_4_5',
src/nidmm/metadata/enums.py:                'name': '_5_5',
src/nidmm/metadata/enums.py:                'name': '_6_5',
src/nidmm/metadata/enums.py:                'name': '_7_5',
src/nidmm/metadata/enums.py:                'name': '_2_WIRE_RES',
src/nidmm/metadata/enums.py:                'name': '_4_WIRE_RES',
src/nidmm/metadata/enums.py:                'name': '_1_MEGAOHM',
src/nidmm/metadata/enums.py:                'name': '_10_MEGAOHM',
src/nidmm/metadata/enums.py:                'name': '_50',
src/nidmm/metadata/enums.py:                'name': '_60',
src/nidmm/metadata/enums.py:                'name': '_44004',
src/nidmm/metadata/enums.py:                'name': '_44006',
src/nidmm/metadata/enums.py:                'name': '_44007',
src/nidmm/metadata/enums.py:                'name': '_2_WIRE_RTD',
src/nidmm/metadata/enums.py:                'name': '_4_WIRE_RTD',
src/nifake/metadata/enums.py:                'name': '_3_5',
src/nifake/metadata/enums.py:                'name': '_4_5',
src/nifake/metadata/enums.py:                'name': '_5_5',
src/nifake/metadata/enums.py:                'name': '_6_5',
src/nifake/metadata/enums.py:                'name': '_7_5',
```

marcoskirsch, 2017-11-09:
Quick look at NI-SCOPE .Net API.

* Tap48Standard 
* Tap48Hanning 
* Tap16Hanning 
...

Fladolcetta, 2017-11-13:
Best to wait on #584

marcoskirsch, 2017-12-19:
Updated list from master:

```
us-aus-ras-130-164-29-12:nimi-python kirsch$ find * | grep generated/ni.*/enums\.py | xargs grep " _"
generated/niscope/enums.py:    _5V_OUT = 13
generated/niscope/enums.py:    _48_TAP_STANDARD = 0
generated/niscope/enums.py:    _48_TAP_HANNING = 1
generated/niscope/enums.py:    _16_TAP_HANNING = 2
generated/niscope/enums.py:    _8_TAP_HANNING = 3
generated/niscope/enums.py:    _1_MEG_OHM = 1000000.0
generated/niscope/enums.py:    _50_OHMS = 50.0
generated/niscope/enums.py:    _480I_59_94_FIELDS_PER_SECOND = 1010
generated/niscope/enums.py:    _480I_60_FIELDS_PER_SECOND = 1011
generated/niscope/enums.py:    _480P_59_94_FRAMES_PER_SECOND = 1015
generated/niscope/enums.py:    _480P_60_FRAMES_PER_SECOND = 1016
generated/niscope/enums.py:    _576I_50_FIELDS_PER_SECOND = 1020
generated/niscope/enums.py:    _576P_50_FRAMES_PER_SECOND = 1025
generated/niscope/enums.py:    _720P_50_FRAMES_PER_SECOND = 1031
generated/niscope/enums.py:    _720P_59_94_FRAMES_PER_SECOND = 1032
generated/niscope/enums.py:    _720P_60_FRAMES_PER_SECOND = 1033
generated/niscope/enums.py:    _1080I_50_FIELDS_PER_SECOND = 1040
generated/niscope/enums.py:    _1080I_59_94_FIELDS_PER_SECOND = 1041
generated/niscope/enums.py:    _1080I_60_FIELDS_PER_SECOND = 1042
generated/niscope/enums.py:    _1080P_24_FRAMES_PER_SECOND = 1045
generated/nidcpower/enums.py:    _50 = 50.0
generated/nidcpower/enums.py:    _60 = 60.0
generated/nidmm/enums.py:    _1_MICROAMP = 1e-06
generated/nidmm/enums.py:    _10_MICROAMP = 1e-05
generated/nidmm/enums.py:    _100_MICROAMP = 0.0001
generated/nidmm/enums.py:    _1_MILLIAMP = 0.001
generated/nidmm/enums.py:    _2_WIRE_RES = 5
generated/nidmm/enums.py:    _4_WIRE_RES = 101
generated/nidmm/enums.py:    _1_MEGAOHM = 1000000.0
generated/nidmm/enums.py:    _10_MEGAOHM = 10000000.0
generated/nidmm/enums.py:    _50 = 50.0
generated/nidmm/enums.py:    _60 = 60.0
generated/nidmm/enums.py:    _44004 = 1
generated/nidmm/enums.py:    _44006 = 2
generated/nidmm/enums.py:    _44007 = 3
generated/nidmm/enums.py:    _2_WIRE_RTD = 3
generated/nidmm/enums.py:    _4_WIRE_RTD = 4
```

marcoskirsch, 2017-12-19:
Based on the grep above, it is apparent that many of these are technically constants, not enums. Specifically the following:

```
generated/niscope/enums.py:    _1_MEG_OHM = 1000000.0
generated/niscope/enums.py:    _50_OHMS = 50.0
generated/nidcpower/enums.py:    _50 = 50.0
generated/nidcpower/enums.py:    _60 = 60.0
generated/nidmm/enums.py:    _1_MICROAMP = 1e-06
generated/nidmm/enums.py:    _10_MICROAMP = 1e-05
generated/nidmm/enums.py:    _100_MICROAMP = 0.0001
generated/nidmm/enums.py:    _1_MILLIAMP = 0.001
generated/nidmm/enums.py:    _1_MEGAOHM = 1000000.0
generated/nidmm/enums.py:    _10_MEGAOHM = 10000000.0
generated/nidmm/enums.py:    _50 = 50.0
generated/nidmm/enums.py:    _60 = 60.0
```

I think we should get rid of those "enums" and just have the client pass the value. Documentation / error reporting are enough. We already did this with NI-DMM's DigitsOfResolution.

marcoskirsch, 2017-12-20:
> Based on the grep above, it is apparent that many of these are technically constants, not enums. 

For the enums that are numeric constants rather than an enum rename problem, I've opened #706. I changed the description of @Fladolcetta 's PR #706 to be about fixing #708.

marcoskirsch, 2018-03-19:
Verified master no longer lists enums that start with `_`

closed 2018-03-19 by marcoskirsch