ni/nimi-python #141: Provide way to lock a session

marcoskirsch, 2017-08-18:
Drivers have LockSession and UnlockSession functions. It allows clients to do multiple calls to the session that are guaranteed to be serialized.

We aren't exposing that in the Python APIs. Perhaps we should. Something like:

with nidmm.Session('dev1') as session:
    spawn_thread_that_uses(session)
    with session.lock()
        # Guaranteed that no other thread can call session in between do_stuff() and do_more_stuff()
        session.do_stuff()
        session.do_more_stuff()

Of course clients have workarounds for this and can do their own synchronization in Python at a higher layer, so I am not sure we should add this functionality.

auchter, 2017-11-30:
> Of course clients have workarounds for this and can do their own synchronization in Python at a higher layer, so I am not sure we should add this functionality.

Unless we're claiming that these APIs aren't thread safe, I think we need to expose the session lock. We also need to make sure that the generated code grabs the session lock when appropriate.

For example, `niscope.Session.fetch` makes a few calls to `niScope_ActualNumWfms` to determine the size of buffers to allocate, and then follows it up with a call to `niScope_Fetch`, but the session lock is not held across that sequence of calls. If another thread increases `fetch_num_records` before the call to `niScope_Fetch`, that's going to result in Fetch stomping on some memory that it shouldn't.

marcoskirsch, 2018-04-30:
* Even when the Python bindings only call one function, in the case of error you call multiple.
* Some generated functions read attributes to get sizes and what not. Are we going to tag those to grab the lock via metadata?
* Python locking mechanism may work. I assume those mutexes aren't recursive so we need to be pretty careful. Only grab in private methods? Do we ever call public methods from other methods?

I think the simplest is to always and blindly grab the IVI session lock. Anything beyond that is optimization.

PR #846 (2018-05-21): Add session lock/unlock
- [X] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).
- [X] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.
- [X] I've added tests applicable for this pull request
### What does this Pull Request accomplish?
* Add `session.lock()` and `session.unlock()`
* `session.lock()` return a context manager that can be used to unlock
### List issues fixed by this Pull Request below, if any.
* Fix #141 

### What testing has been done?
* Unit
* System
closed 2018-05-21 by marcoskirsch