ni/nimi-python #152: NI-DMM System tests fail on a simulated PCI-4070

Fladolcetta, 2017-08-23:
- test_get_dev_temp returns 0.0 and the test is looking for a range of 20-40
- test_multi_point_acquisition fails due to max time exceeded before operation completed on read_multi_point call.
- test_method_with_enum_output_type_method fails as "assert <AcquisitionStatus.NO_ACQUISITION_IN_PROGRESS: 4> == 4"

marcoskirsch, 2017-08-23:
> test_get_dev_temp  returns 0.0 and the test is looking for a range of 20-40

I doubt this is the case on 408x.

> test_multi_point_acquisition fails due to max time exceeded before operation completed on read_multi_point call.

I doubt this is the case on 408x.

> test_method_with_enum_output_type_method fails as "assert <AcquisitionStatus.NO_ACQUISITION_IN_PROGRESS: 4> == 4"

I doubt this is the case on 408x.

injaleea, 2017-08-24:
> test_get_dev_temp returns 0.0 and the test is looking for a range of 20-40

Simulated device always returned dev_temp as 0. right ? [ LabVIEW returns dev_temp = 0 for simulated devices ]

> test_multi_point_acquisition fails due to max time exceeded before operation completed on read_multi_point call.

Ideally **Maximum number of measurements = Trigger Count * Sample Count**. 
But for Simulated devices ,it looks like trigger happens only once ,
**ie,
Maximum number of measurement = Sample Count** [Noticed same behavior with LabVIEW APIs aswell. Should it be an issue on LabVIEW APIs aswell?] 

> test_method_with_enum_output_type_method fails as "assert <AcquisitionStatus.NO_ACQUISITION_IN_PROGRESS: 4> == 4"

https://github.com/ni/nimi-python/pull/151 should have fix for this issue

marcoskirsch, 2017-08-24:
```
Simulated device always returned dev_temp as 0. right ?
[ LabVIEW returns dev_temp = 0 for simulated devices ]
```

It depends which device you are simulating.

```
But for Simulated devices ,it looks like trigger happens only once ,ie,
Maximum number of measurement = Sample Count
```

This also depends on which device you're simulating. I'm fairly confident you "receive" all triggers when you simulate a 408x. Not sure about others.

`[Noticed same behavior with LabVIEW APIs aswell. Should it be an issue on LabVIEW APIs aswell?]`

To be clear, the LabVIEW API, just like the Python bindings and the .NET wrappers simply call into the C API. Behaviors should be identical with few exceptions because they all call the same exact functions.

PR #194 (2017-08-29): Updating system tests to use a simulated session instead of actual ha…
…rdware.

[x ] This contribution adheres to [CONTRIBUTING.md]

### What does this Pull Request accomplish?

Fixes #192 
Fixes #152 

### Why should this Pull Request be merged?

This prevents failures / skipped tests when running system tests.
It also allows system tests to be run without hardware.

### What testing has been done?

Built, ran flake8, and ran system tests
closed 2017-08-29 by marcoskirsch