ni/nimi-python #172: make produces distracting build warnings.

marcoskirsch, 2017-08-28:
From clean:

```
Kirsch0:nimi-python kirsch$ make

Making nifake
     Making dir: bin/nifake
     Making dir: bin/nifake/nifake
     Making dir: bin/nifake/nifake/tests
     Making dir: bin/nifake/log
     Making dir: bin/nifake/system_tests
     Making dir: bin/nifake/examples
     Generating: bin/nifake/nifake/enums.py
     Generating: bin/nifake/nifake/library.py
     Generating: bin/nifake/nifake/session.py
     Generating: bin/nifake/nifake/errors.py
     Generating: bin/nifake/nifake/ctypes_library.py
     Generating: bin/nifake/nifake/tests/mock_helper.py
     Generating: bin/nifake/nifake/__init__.py
        Copying: bin/nifake/nifake/ctypes_types.py
        Copying: bin/nifake/nifake/python_types.py
     Generating: docs/nifake/session.rst
     Generating: docs/nifake/enums.rst
     Generating: docs/nifake/attributes.rst
     Generating: docs/nifake/functions.rst
        Copying: bin/nifake/nifake/tests/test_session.py
     Generating: bin/nifake/setup.py
       Updating: generated/nifake/
        Copying: bin/nifake/README.rst
 Running pytest: bin/nifake/log/tests_passed
 Creating wheel: bin/nifake/dist
zip_safe flag not set; analyzing archive contents...
 Creating sdist: bin/nifake/dist

Making nidmm
     Making dir: bin/nidmm
     Making dir: bin/nidmm/nidmm
     Making dir: bin/nidmm/nidmm/tests
     Making dir: bin/nidmm/log
     Making dir: bin/nidmm/system_tests
     Making dir: bin/nidmm/examples
     Generating: bin/nidmm/nidmm/enums.py
     Generating: bin/nidmm/nidmm/library.py
     Generating: bin/nidmm/nidmm/session.py
     Generating: bin/nidmm/nidmm/errors.py
     Generating: bin/nidmm/nidmm/ctypes_library.py
     Generating: bin/nidmm/nidmm/tests/mock_helper.py
     Generating: bin/nidmm/nidmm/__init__.py
        Copying: bin/nidmm/nidmm/ctypes_types.py
        Copying: bin/nidmm/nidmm/python_types.py
     Generating: docs/nidmm/session.rst
     Generating: docs/nidmm/enums.rst
     Generating: docs/nidmm/attributes.rst
     Generating: docs/nidmm/functions.rst
        Copying: bin/nidmm/nidmm/tests/test_session.py
     Generating: bin/nidmm/setup.py
       Updating: generated/nidmm/
        Copying: bin/nidmm/system_tests/test_system_nidmm.py
        Copying: bin/nidmm/examples/nidmm_measurement.py
        Copying: bin/nidmm/README.rst
 Running pytest: bin/nidmm/log/tests_passed
 Creating wheel: bin/nidmm/dist
zip_safe flag not set; analyzing archive contents...
 Creating sdist: bin/nidmm/dist

Making nimodinst
     Making dir: bin/nimodinst
     Making dir: bin/nimodinst/nimodinst
     Making dir: bin/nimodinst/nimodinst/tests
     Making dir: bin/nimodinst/log
     Making dir: bin/nimodinst/system_tests
     Making dir: bin/nimodinst/examples
     Generating: bin/nimodinst/nimodinst/library.py
     Generating: bin/nimodinst/nimodinst/session.py
     Generating: bin/nimodinst/nimodinst/errors.py
     Generating: bin/nimodinst/nimodinst/ctypes_library.py
     Generating: bin/nimodinst/nimodinst/tests/mock_helper.py
     Generating: bin/nimodinst/nimodinst/__init__.py
        Copying: bin/nimodinst/nimodinst/ctypes_types.py
        Copying: bin/nimodinst/nimodinst/python_types.py
     Generating: docs/nimodinst/session.rst
     Generating: docs/nimodinst/attributes.rst
     Generating: docs/nimodinst/functions.rst
        Copying: bin/nimodinst/nimodinst/tests/test_modinst.py
     Generating: bin/nimodinst/setup.py
       Updating: generated/nimodinst/
        Copying: bin/nimodinst/system_tests/test_system_nimodinst.py
        Copying: bin/nimodinst/examples/nimodinst_all_devices.py
        Copying: bin/nimodinst/README.rst
 Running pytest: bin/nimodinst/log/tests_passed
 Creating wheel: bin/nimodinst/dist
zip_safe flag not set; analyzing archive contents...
 Creating sdist: bin/nimodinst/dist
```

Produces annoying:

`zip_safe flag not set; analyzing archive contents...`

Incremental builds:

```
Kirsch0:nimi-python kirsch$ make

Making nifake
make[1]: Nothing to be done for `MKDIR'.
       Updating: generated/nifake/
 Creating wheel: bin/nifake/dist
 Creating sdist: bin/nifake/dist

Making nidmm
make[1]: Nothing to be done for `MKDIR'.
       Updating: generated/nidmm/
 Creating wheel: bin/nidmm/dist
 Creating sdist: bin/nidmm/dist

Making nimodinst
make[1]: Nothing to be done for `MKDIR'.
       Updating: generated/nimodinst/
 Creating wheel: bin/nimodinst/dist
 Creating sdist: bin/nimodinst/dist
```

Produces annoying:

`make[1]: Nothing to be done for `MKDIR'.`

And make clean:

```
Kirsch0:nimi-python kirsch$ make clean
Cleaning...
make: [clean] Error 1 (ignored)
```

Produces annoying:

`make: [clean] Error 1 (ignored)`

texasaggie97, 2017-08-29:
Now that rebuilding when not required has been cleaned up, `make` after a `make clean && make` will give this:
```
Making nifake

Making nidmm

Making nimodinst
```

Is it acceptable to not say anything?

marcoskirsch, 2017-08-29:
> Now that rebuilding when not required has been cleaned up

What is _now_? can you link to the PR?

> Is it acceptable to not say anything?

On my system, when I make a second time from master:

```
Kirsch0:nimi-python kirsch$ make

Making nifake
make[1]: Nothing to be done for `MKDIR'.
make[1]: Nothing to be done for `all'.

Making nidmm
make[1]: Nothing to be done for `MKDIR'.
make[1]: Nothing to be done for `all'.

Making nimodinst
make[1]: Nothing to be done for `MKDIR'.
make[1]: Nothing to be done for `all'.
Kirsch0:nimi-python kirsch$ 
```

I think the MKDIR is annoying.
I think the 'all' thing is nice.
We can live with it as it stands, though.

PR #196 (2017-08-29): Clean up build output
[X] This contribution adheres to [CONTRIBUTING.md]
### What does this Pull Request accomplish?
* Fixes #172 
* Suppress problematic make output
* Set zip_safe=True to remove warning

### Why should this Pull Request be merged?
* Fixes issue

### What testing has been done?
* Travis
closed 2017-08-29 by marcoskirsch