ni/nimi-python #187: Add channels for properties on session object

Fladolcetta, 2017-08-28:
Example:

`session.voltage_level[‘0,1’] = 5 `

marcoskirsch, 2017-09-12:
There are some implementation challenges to this. But more importantly, I'm not convinced this is something we want to do, even though it's "clever" in a Modular Instruments specific clever way.

From a semantic point of view, consider the following:

```
session.voltage_level = 5
level = session.voltage_level
```

The first line sets the voltage_level attribute on the session to the integer 5, using '' as the channel which in our APIs means "all the channels in the session".
The second line gets the voltage_level attribute from the session using channel '' and assigns it to level variable which now contains an integer value.

But we also want this to work, according to this issue:

```
session.voltage_level['2'] = 5
level2 = session.voltage_level['2']
```

We would be making voltage_level property act as an integer but also as a mapping type. I don't think this is Pythonic, and it only makes sense to our brain that have been immersed in the "Weird World of MI®" for  far too long.

Feedback is welcome.

thejcannon, 2017-09-12:
Does the metadata know which attributes are session-bound and which are channel-bound?
It makes me wonder if we should only allow session-bound attributes to be accessed from the session instance itself:
```
print(session.alias)
```
The session instance also acts as a container of channels:
```
for channel in session:
   channel.voltage_level = 5

session[channel_index].voltage_level = 3
```

But trying to access channel-based attributes on the session without getting the channel object would result in an error. (Hopefully with a helpful message).

epage, 2017-09-12:
I think the approach that DAQmx took to solve this problem would work well here.

```python
session.channels.all().voltage_level = 5
level = session.channels.all().voltage_level

session.channels[2].voltage_level = 5
level = session.channels[2].voltage_level

session.channels['0:2'].voltage_level = 5
level = session.channels['0:2'].voltage_level

session.channels['0, 1, 2'].voltage_level = 5
level = session.channels['0, 1, 2'].voltage_level

levels = [ch.voltage_level for ch in session.channels]

chans = session.channels['0:2'] + session.channels['5:10'])
chans.voltage_level
```

- `session.channels` is called a [Channel Collection](https://github.com/ni/nidaqmx-python/blob/master/nidaqmx/_task_modules/channel_collection.py) (see [ai](https://github.com/ni/nidaqmx-python/blob/master/nidaqmx/_task_modules/ai_channel_collection.py))
  - performs mapping
  - owns collection manipulation (like adding channels)
- `session.channels.all()` returns a [Channel](https://github.com/ni/nidaqmx-python/blob/master/nidaqmx/_task_modules/channels/channel.py) object (see [ai](https://github.com/ni/nidaqmx-python/blob/master/nidaqmx/_task_modules/channels/ai_channel.py)).
  - Owns the properties
  - Owns any single-channel or channel subsection functions (like save channel)

In MI's case, I doubt you have the whole-collection or single-channel manipulation like, so you might be able to get away with merging the two objects.  The only reason I can think of to keep them separate is consistency with DAQmx which does have some value.

thejcannon, 2017-09-12:
I wouldn't mind having channels being an property of the session instance itself. It definitely is more explicit.
My recommendation would be to try and make it a simple list of Channel objects (not some user-defined type). That way clients get the familiar syntax of a list: I.e. `chans = session.channels[0:2] +session.channels[5:10]`. And can easily iterate over it.

epage, 2017-09-12:
> My recommendation would be to try and make it a simple list of Channel objects (not some user-defined type). That way clients get the familiar syntax of a list: I.e. chans = session.channels[0:2] +session.channels[5:10]. And can easily iterate over it.

1. I imagine the most common case is to set all channels the same.  A list of channel objects is more cumbersome for that
2. As I've spent more time considering the API design choice of DAQmx, I think its Structs-of-Arrays approach it takes is much better than the standard OOP Arrays-of-Structs approach for expressing ideas.
3. When needing to deal with individual channels, its still pretty straightforward to iterate on them and deal with them.

MrElusive, 2017-09-12:
> > My recommendation would be to try and make it a simple list of Channel objects (not some user-defined type). That way clients get the familiar syntax of a list: I.e. chans = session.channels[0:2] +session.channels[5:10]. And can easily iterate over it.

> I imagine the most common case is to set all channels the same. A list of channel objects is more cumbersome for that

I agree. The channels object would be better off as an object with an overridden __setitem__ and __iter__ dunders rather than a simple list. This would allow for setting all the channels at the same time, while providing the familiar syntax of a list. Additionally it would allow for two types of channel selection modes:

For a session that contains all even channels on a 24 channel device, the following calls would be the same:
```python
session.channels["0,2,4,6,8"]
session.channels[0:4]
```

This is also the approach used in the .NET API:
```c#
                dcPowerSession.Outputs[ChannelName].Source.Output.Function = DCPowerSourceOutputFunction.DCVoltage;
                dcPowerSession.Outputs[ChannelName].Source.Voltage.VoltageLevel = VoltageLevel1;
                dcPowerSession.Outputs[ChannelName].Source.Voltage.CurrentLimit = CurrentLimit;
                dcPowerSession.Outputs[ChannelName].Source.Voltage.VoltageLevelRange = VoltageLevelRange;
                dcPowerSession.Outputs[ChannelName].Source.Voltage.CurrentLimitRange = CurrentLimitRange;
```

MrElusive, 2017-09-12:
> Example:
> 
> session.voltage_level[‘0,1’] = 5
> 
> Would set voltage_level property to 5 on channels 0, 1.

> There are some implementation challenges to this. But more importantly, I'm not convinced this is something we want to do, even though it's "clever" in a Modular Instruments specific clever way.

Indeed, having tried to implement this approach and observing the technical difficulties involved, it quickly begins to smell of a design flaw. I think the precise issue is what Marcos pointed out: session properties should not represent both mapping types and driver attributes. This would be a strong violation of separation of concerns.

epage, 2017-09-12:
Tangent: So it looks like the .NET API mapped subobjects for every menu level in the property node?  That seems excessive.  I like the balance in the DAQmx .NET and python APIs where it is a sub object per property node (LabVIEW) or filter (LabVIEW NXG) but within that level, everything is flat.  Strikes a balance between flat and deeply nested.

MrElusive, 2017-09-12:
I also remember seeing an example of using context managers for channel selection (was that from you, Marcos?):

```python
with session.channel[0-3] as channels:
    channels.voltage_level
    ...

with session.channel[0-3]:
    session.voltage_level
    ...
```

My preference would be to avoid adding another context manager to avoid excessive indentation levels:

```python
with open("myMeasurementResultsFile.csv", "w") as measurementResults:
    with NIDCPowerSession() as session:
        with session.initiate():
            for i in range(1, 10):
                with session.channels[0-3]:
                    measurementsResults.write(session.measure(voltage))            
```

Of course, it doesn't hurt to add the context manager and it doesn't mean clients have to use it. However, I don't think it adds value the same way that providing a context manager for the session itself does (there is no initialize/close analog for channels).

MrElusive, 2017-09-12:
> Tangent: So it looks like the .NET API mapped subobjects for every menu level in the property node? That seems excessive. I like the balance in the DAQmx .NET and python APIs where it is a sub [...]

What makes it worse is that the .NET API mapping is not one to one. This is no Source -> Output -> Function hierarchy: there is simply Source -> Output Function

So, while the python API will diverge from the .NET API in terms of organization style, I think it's desirable.

marcoskirsch, 2017-09-12:
> Does the metadata know which attributes are session-bound and which are channel-bound?

It does. But I'm very hesitant to use that. For one, it's very fluid: for example, NI-DCPower is inconsistent across different devices - or at the very least attributes that used to be "device attributes" were converted to "channel attributes".

> session[channel_index].voltage_level = 3

This is too much like the original problem. See @MrElusive 's comment "session properties should not represent both mapping types and driver attributes".

> I also remember seeing an example of using context managers for channel selection (was that from you, Marcos?):

Yes, that was me. As you say, it would be optional to use. You can do either:

    with Session.channel("0, 1") as chan_mgr:
        chan_mgt.voltage_level = 5

or 

    Session.channel("0,1").voltage_level = 5

I don't see any harm allowing this.

> I think the approach that DAQmx took to solve this problem would work well here.

I like the consistency and the fact that I can blame nidaqmx for making the wrong choice if it turns out to be the wrong choice :)

Lots of great feedback. As for the original question related to this issue, so far everyone appears to agree that we probably should close as "won't fix". Anyone disagree?

PR #334 (2017-09-20): Support for channel based properties
[X] This contribution adheres to [CONTRIBUTING.md]

### What does this Pull Request accomplish?

* Adds API support for setting and getting channel-based properties.
* Paves the way for how we want to handle the channel parameter from out C API.
* Fixes #185 
* Fixes #187 

Channel based properties can be used in two ways:

```
# sets voltage level in channel 2 to 5 volts.
session['2'].voltage_level = 5
session['2']. output_enabled = 5
```

```
# using a context manager.
with session['2'] as c2:
    c2.voltage_level = 5
    c2.output_enabled = True
```

The [... cut]
closed 2017-09-20 by marcoskirsch