ni/nimi-python #188: Ability to use integer / string / list / range as channel

Fladolcetta, 2017-08-28:


MrElusive, 2017-09-20:
@epage 
@marcoskirsch 

If I'm not mistaken, using a list/range will have a different behavior than using a string, right? With a string, a single driver call is made. But with list/range, the session will iterate and make a separate driver call for each channel (unless a string is reconstructed from the list/range). Is this how it works for DAQmx?

epage, 2017-09-20:
nidaqmx does one call to find the channel names, applies the index/slice on that, and then makes the actual property call once with that.

See https://github.com/ni/nidaqmx-python/blob/master/nidaqmx/_task_modules/channel_collection.py#L43

MrElusive, 2017-09-20:
Ah, I see. DCPower doesn't have the equivalent of DAQmxGetTaskChannels, so we'll need to get that exposed first.

epage, 2017-09-20:
I assume thats because your Sessions include all channels?  So you could lazily look up what physical channels are associated with the Session's board and cache it and just use that?

I did something like that for [nixnet](https://github.com/ni/nixnet-python/blob/master/nixnet/_session/collection.py#L14).

MrElusive, 2017-09-21:
No, each session can be bound to a subset of channels; we just haven't exposed a mechanism to easily retrieve that information. Of course, there is the inherent niDCPower_GetChannelName which, like you said, could be called lazily and cached.

```python
def get_channel_name(one_based_index):
    buffer_size = dcpower_lib.get_channel_name(one_based_index, 0, None)
    channel_name_buffer = create_string_buffer(buffer_size)
    dcpower_lib.get_channel_name(one_based_index, buffer_size, channel_name_buffer)
    return channel_name_buffer.value.decode("ASCII")

def get_channel_count():
    channel_count = c_uint32_t()
    dcpower_lib.parse_channel_count(session_channels, byref(channel_count))
    return channel_count

channel_names = [get_channel_name(index + 1) for index in range(get_channel_count())]
```

epage, 2017-09-21:
It is static though for the lifetime of a session?  Thats a big help.

marcoskirsch, 2017-09-22:
> If I'm not mistaken, using a list/range will have a different behavior than using a string, right? With a string, a single driver call is made.

The idea is to convert whatever is passed in into a string suitable for the driver to consume, so a single driver call is still made.

> Ah, I see. DCPower doesn't have the equivalent of DAQmxGetTaskChannels, so we'll need to get that exposed first.

Anything requiring C API changes would be outside of the scope of nimi-pyhton 1.0, but can be considered for the future.

> Of course, there is the inherent niDCPower_GetChannelName which, like you said, could be called lazily and cached.

That's what we were thinking.

> It is static though for the lifetime of a session? Thats a big help.

Yep.

marcoskirsch, 2019-03-07:
This was fixed well before 1.0.

closed 2019-03-07 by marcoskirsch