ni/nimi-python #234: Not having documentation in functions.py causes flake8 failures

Fladolcetta, 2017-09-05:
If a function in functions.py does not have documentation, our build adds the following:

        '''self_test

        

        Returns:
            self_test_result (int):
            self_test_message (int):
        '''

Notice Line3 above has a bunch of spaces added where documentation would normally go. This causes a flake8 failure as shown below:

.\niswitch\niswitch\session.py:1721:1: W293 blank line contains whitespace
        '''self_test

        Returns:
            self_test_result (int):
            self_test_message (int):
        '''

marcoskirsch, 2017-09-05:
I think this issue, if I understand correctly, should be renamed: "Generated code fails flake8 when function metadata is missing documentation".

Whether this is valid input to the code-generator or not needs to be discussed... maybe it's not so it's not something we want to fix.

Fladolcetta, 2017-09-25:
Our metadata should always have documentation.

closed 2017-09-25 by Fladolcetta