ni/nimi-python #248: Incorporate NI-DMM system tests into NI internal Automated Test Suite

marcoskirsch, 2017-09-05:
NI maintains an Automated Test Suite for the driver.
It needs to install the corresponding nimi-python package and run the corresponding system tests as part of its execution.

This way we can ensure that future versions of the driver work with nimi-python.

Fladolcetta, 2017-09-27:
Expected timeline ~Oct. 6th 2017

injaleea, 2017-10-03:
Currently we system tests are written for a simulated 4080 device. Do we need to run system tests only for this simulated device or will have to run for all available devices in testsuite?

> Expected timeline ~Oct. 6th 2017
@Fladolcetta : we will be adding NI-DMM system tests into NI internal Automated Test Suite once we have nidmm_system_tests tox environment available. We may have to re-consider deadline based on tox environments availability if we are not getting it pulled before 04th of Oct.

Fladolcetta, 2017-10-03:
We do not required system tests to be device specific. We should continue using the simulated 408x devices for system tests.

The timeline has been adjusted to release DCPower on 10/13 due to various technical debt items.

Fladolcetta, 2017-11-06:
Added to test suites

closed 2017-11-06 by Fladolcetta