ni/nimi-python #268: niswitch system test test_enum_attribute causes access violation

texasaggie97, 2017-09-07:
niswitch system test test_enum_attribute causes access violation on the system test runner

This test is commented out for now

marcoskirsch, 2017-09-12:
On my test machine (Windows 7 64-bit) I ran the following code:

```
import niswitch

with niswitch.Session('', False, True, 'Simulate=1, DriverSetup=topology:2532/1-Wire 4x128 Matrix') as session:
    assert session.scan_mode == niswitch.ScanMode.BREAK_BEFORE_MAKE
```

When executed in Python 3.6.2 it works.
When executed in Python 2.7.13 it hangs.

@Fladolcetta to confirm that the same is the case on nimi-bot VM.

marcoskirsch, 2017-09-12:
I can confirm that if I kill the Python process while it's opening a simulated session to the DAQmx switch, the system is now in a bad state. Restarting mxssvr puts me back in a good state.

I'm using a small example that runs things in a loop and prints the time:

```
import niswitch
import time

i = 0
while True:
    start = time.time()

    with niswitch.Session('', False, True, 'Simulate=1, DriverSetup=topology:2532/1-Wire 4x128 Matrix') as session:
        assert session.scan_mode == niswitch.ScanMode.BREAK_BEFORE_MAKE
    i = i + 1
    end = time.time()
    print("Iteration {0}: {1:06.3f} seconds".format(i, (end-start), " seconds"))
```

marcoskirsch, 2017-09-14:
We've determined that this is an NI-SWITCH bug with DAQmx-based devices.
We have worked around it in the test, and filed NI internal bug 666018.

PR #276 (2017-09-14): Updating system tests to re-add the enum attribute test
[x ] This contribution adheres to [CONTRIBUTING.md]

### What does this Pull Request accomplish?

Re-adds the system test that was crashing nimi-bot (system test machine).
The system test has been modified to disable reset which worked as a workaround. This appears to be an internal NI-SWITCH bug with DAQmx based devices. Filed as NI internal bug 666018.

### Why should this Pull Request be merged?

Fixes #268 
Improves test coverage.

### What testing has been done?

Ran system tests, verified system test machine test passed
closed 2017-09-14 by marcoskirsch