ni/nimi-python #286: Stop copying system tests to bin/

marcoskirsch, 2017-09-10:
We copy system tests at build time, from:

    src/niswitch/system_tests/test_system_niswitch.py

to:

    bin/niswitch/system_tests/test_system_niswitch.py

If memory serves me right, we do this for historical reasons: we used to play import path tricks rather than install the modules so we needed the tests to be in a specific location relative to the built modules. This is no longer true - now we install the modules in the virtual environment. So we should be able to stop copying the files and run them from their location in src/.

texasaggie97, 2017-09-11:
We also copy them there so the flake8 tests run on them.

Should we try to change the flake8 tests to test them in src as well?

marcoskirsch, 2017-09-11:
> Should we try to change the flake8 tests to test them in src as well?

Yeah, unless it poses a technical challenge.

PR #309 (2017-09-14): Don't copy system tests or examples to bin/ folder
[X] This contribution adheres to [CONTRIBUTING.md]
### What does this Pull Request accomplish?
* Fixes #286 
* Don't copy system tests or examples to bin/
* Run flake8 tests on them in src/\<driver\>/\[system_tests|examples\]

### Why should this Pull Request be merged?
* Fixes issue

### What testing has been done?
* Travis & system tests
closed 2017-09-14 by marcoskirsch