ni/nimi-python #293: Add NI-TClk Support

Fladolcetta, 2017-09-11:
Add NI-TClk support.

marcoskirsch, 2017-10-23:
We've decided not to do this for the 0.9 December 2017 release. Will close issue.
If there's customer demand for an nitclk module, we'll have to add it at that point.

marcoskirsch, 2018-11-01:
> If the driver supports NI-TClk and nitclk is installed, add the attributes to the driver session

This is a very interesting idea. There is a bit of "surprise" / "side effect" / "clever" factor to this that makes me feel a bit uncomfortable. If we do go ahead with it, we should consider grouping the NI-TClk attributes in a container: `scope.nitclk.sync_pulse_source = 'PXI_Trig0'`

> This will allow us to use the already existing repeated capabilities mechanism

Do you mean something like this? `niscope.channels[0].nitclks.sync_pulse_source = 'PXI_Trig0`

auchter, 2018-11-01:
> Even though the nitclk.Session() doesn't need to call a close or do any other cleanup, it would still be a context manager.

Why make it a context manager then?

The example provided makes it look like there's something special about the TClk context. I'm concerned that this is going to mislead users into thinking that the context implies being synchronized, and that leaving the context will cause the devices to be "unsynchronized" and revert back to their pre-synchronization state. But that's not the case: the devices will still be synchronized upon leaving that context, the trigger resynchronizers will still be configured for a TClk'd acquisition, etc.

epage, 2018-11-01:
> If the driver supports NI-TClk and nitclk is installed, add the attributes to the driver session

My default is to be against dynamic functionality but this *might* be a case where it makes sense.  I do agree with @marcoskirsch about having it in a dedicated container.

auchter, 2018-11-01:
>  If the driver supports NI-TClk and nitclk is installed, add the attributes to the driver session

I agree with both @marcoskirsch and @epage.

Please consider how Python APIs for drivers that aren't in this repository will implement this. I wouldn't be in favor of this approach if it's implemented as part of the codegeneration of each driver API. However, if all of the attribute implementation lives in an `nitclk` module, such that driver APIs can implement this along the lines of:
```python
@property
def tclk(self):
    import nitclk
    return nitclk.properties(self.get_session_reference())
```
then I think this is a reasonable approach.

marcoskirsch, 2018-11-01:
> Please consider how Python APIs for drivers that aren't in this repository will implement this.

@auchter I assume you are referring to [nifpga-python](https://github.com/ni/nifpga-python). Is there anything else at the moment?

We'd want to make sure that the nitclk module sourced here will work with that API.

We also should make sure that sourcing nitclk with the rest of nimi-python makes sense.

texasaggie97, 2018-11-02:
I updated the issue text:
1. Properties get added to `tclk` container rather than directly to `Session`
1. Add `tclk` property snippet - `properties` not private though since there may be other drivers that need to access it
1. No context manager - strikethrough
1. Recieved information offline about one of the assumptions - strikethrough
1. Updated example

texasaggie97, 2018-11-05:
One more area for discussion: Should we just add a dependency on nitclk from the other drivers and assume it is installed?

As far as I can tell, all other languages/ADEs handle NI-TClk this way. If the driver support is installed for the given language/ADE, then NI-TClk support is also installed.

Now, just because we "have always done it this way" doesn't automatically mean we need to here, but it is a precedent.

epage, 2018-11-05:
> As far as I can tell, all other languages/ADEs handle NI-TClk this way. If the driver support is installed for the given language/ADE, then NI-TClk support is also installed.

In my mind, this is an unfortunate thing and not an ideal.  We should be moving towards software integration pieces being add ons rather than required.

marcoskirsch, 2018-11-05:
> As far as I can tell, all other languages/ADEs handle NI-TClk this way. If the driver support is installed for the given language/ADE, then NI-TClk support is also installed.

This is true for NI-ModInst as well, except in the Python case.

bjmuld, 2018-11-28:
I see this issue was active fairly recently... can you guys comment on the status at present?  What timescale are you thinking for a usable package?

marcoskirsch, 2018-11-28:
We are debating priorities internally.

Some favor work on internal improvements to the code to make future work easier.
Some believe #940 should come first.
Some believe #293 (this) should be our next large deliverable.

User feedback can influence things. Can you provide information

* What you're trying to do and why?
* How much pain you're willing to endure if we put out something to get you unstuck but far from final – which would require retrofits as we refine things?
* What is your project timeframe?
* Would you be willing to help out in development of the nitclk module?

Thanks!

bjmuld, 2018-11-28:
> What you're trying to do and why? Maybe old-school synchronization using triggers is good enough?

I'm researcher who has an automated experimentation platform for circuits implemented in python. Live mostly in python.  I do (try to do) synchronous generation and capture across a couple PXI modules).

> How much pain you're willing to endure if we put out something to get you unstuck but far from final – which would require retrofits as we refine things?

TBH, I probably wouldn't notice the pain (as far as NI-Tclk is concerned) provided my setups still run. Meaning, a sub-optimal synchronization would be tolerated.

> What is your project timeframe?

I'm going to defend my diss. in Jan, God-willing, so will be running all experiments btw now and then.

> Would you be willing to help out in development of the nitclk module?

Perhaps.  But probably not in any significant way.  Austin sounds nice, tho :)

marcoskirsch, 2018-11-28:
> I'm researcher who has an automated experimentation platform for circuits implemented in python. Live mostly in python. I do (try to do) synchronous generation and capture across a couple PXI modules).

Are these digitizers? Waveform generators? A mix?

> Meaning, a sub-optimal synchronization would be tolerated.

This is what I was trying to get at with "Maybe old-school synchronization using triggers is good enough". With NI-TClk you get nanosecond-level synchronization. But you can still use triggers to synchronize your devices without using NI-TClk. Your devices will be synchronized within a sample clock period but you may have some jitter. If this is acceptable for your application then you don't need NI-TClk - rather you'd be settting up some trigger routes using the device's API. If that's the case and you need help with it, then it becomes a [regular NI support question](ni.com/support) which is a good thing as you would not be tied to nimi-python development schedule.

bjmuld, 2018-11-28:
> Are these digitizers? Waveform generators? A mix?

A mix.

> If that's the case ... you would not be tied to nimi-python development schedule.

Doesn't sound like nimi-python/tclk is in the cards for that reason.

marcoskirsch, 2018-11-28:
> Doesn't sound like nimi-python/tclk is in the cards for that reason.

I wouldn't bet on nitclk being ready by January, and if you don't need NI-TClk-level synchronization for your application, I would suggest you don't add this risk to your work and simply set up trigger synchronization yourself.

bjmuld, 2018-11-29:
Roger that.  thanks!

PR #1054 (2019-10-21): Enable driver support for nitclk
- [X] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).
- [X] ~~I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.~~
- [X] I've added tests applicable for this pull request
### What does this Pull Request accomplish?
* Drivers can now opt into nitclk support
* nifake test for correct type
* `nitclk` is now ready for use

### List issues fixed by this Pull Request below, if any.
* Fix #293 
### What testing has been done?
* Unit
* System
closed 2019-10-21 by marcoskirsch