ni/nimi-python #318: Metadata names are extracted and this is brittle for Python APIs

marcoskirsch, 2017-09-14:
We use internal NI tools to extract API metadata from two sources:

1. Attribute and enum metadata is extracted from LabVIEW property nodes.
2. Function metadata is extracted from CVI FP file.

Theses are better sources than the C header file because they contain more information such as enums and documentation.

However, there is a big risk: Internal NI developers can change names in LabVIEW property nodes without breaking LabVIEW clients. They can also change parameter names in CVI without breaking CVI clients. There's a big risk that these changes are made and accidentally propagated to Python APIs where they _will_ be source breakers from clients.

To make things worse, MI APIs in general aren't great with their naming consistency especially in CVI/H files. You will see things like numBytes, size, bufferSize, buffSize, numberOfButes, bufferSizeInBytes, etc. used pretty randomly. This is terrible for the Python experience where client may type parameter names, and couples us even more.

In order to minimize this risk, it appears internal NI developers need to:

* comb through these sources of metadata and get them to a more consistent state
* establish internal processes so that breakages are caught in an automatic and timely fashion

Fladolcetta, 2017-11-06:
Given that we are now basing the names of the attributes and the enum values off of the C API the only thing that is still brittle is function parameter names.

MrDuff8, 2019-07-25:
NI internal tools were overhauled, so this is no longer considered an issue.

closed 2019-07-25 by MrDuff8