ni/nimi-python #320: In niFake_Read(), when there is a warning, return value is 0, and not what we specify

bhaswath, 2017-09-15:
Tested with the following code. Getting AssertionError, value returned in session.read(test_maximum_time) is 0.0 and not nan.

    def test_read_with_warning(self):
        test_maximum_time = 10
        test_reading = float('nan')
        test_error_code = 42
        test_error_desc = "The answer to the ultimate question, only positive"
        self.patched_library.niFake_Read.side_effect = self.side_effects_helper.niFake_Read
        self.side_effects_helper['Read']['return'] = test_error_code
        self.side_effects_helper['Read']['reading'] = test_reading
        self.patched_library.niFake_GetError.side_effect = self.side_effects_helper.niFake_GetError
        self.side_effects_helper['GetError']['errorCode'] = test_error_code
        self.side_effects_helper['GetError']['description'] = test_error_desc
        with nifake.Session('dev1') as session:
            with warnings.catch_warnings(record=True) as w:
                assert test_reading == session.read(test_maximum_time)
                assert len(w) == 1
                assert issubclass(w[0].category, nifake.NifakeWarning)
                assert test_error_desc in str(w[0].message)

marcoskirsch, 2017-09-15:
Did you try reproducing this with hardware?

bhaswath, 2017-09-19:
The NI-DMM Example will return nan along with the overrange warning, as expected.

marcoskirsch, 2017-09-19:
Great, I think I know where the bug is: mock_helper.py returns immediately when return code is programmed to be anything other than 0. Please do PR with your test commented out, we can fix this today.

marcoskirsch, 2017-09-20:
See discussion in PR, this may not be as simple a fix as I initially thought.
Of course there's always the option of not using mock_helper and setting all side effects in the unit test directly.

Fladolcetta, 2017-10-17:
This is not an issue, and is a misuse of the mock framework. To get the behavior we need from the mock library, we write our own side effect rather than relying on the default side effect from mock_helper.py

commit 69d1a7f1 (2017-09-19): Fix #320: Do side effects in the warning case.
PR #470 (2017-10-17): Add more functions and unit tests to get cover todos in niFake
[ x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

[ x] I've added any unit tests that are applicable for this pull request

### What does this Pull Request accomplish?

Updates generator so error on close raises an exception

Adds Unit Tests to cover todo items test_session.py

- test_close_with_error
- test_session_context_manager_init_with_error
- test_session_context_manager_close_with_error
- test_enum_input_function_with_defaults_bad_type_error
- test_read_with [... cut]
closed 2017-10-17 by Fladolcetta