ni/nimi-python #343: Channel property access through context manager is probably an abuse of context managers

epage, 2017-09-20:
#334 added context managers for accessing channel properties
```python
# using a context manager.
with session['2'] as c2:
    c2.voltage_level = 5
    c2.output_enabled = True
```

Context managers are generally applied when you want to have code execute at the end of a block, especially if its symmetrical (open / close).  One less common use I admittedly make for context managers is to make my code structure look like my XML structure.

This use of a context manager though is more equivalent of VB's [with block](https://msdn.microsoft.com/en-us/library/wc500chb(v=vs.90).aspx) and provides no value over assigning to a variable.
- A variable needs to be created anyways
- with-statements do not create a new scope and the variable created from one will be available outside of the with-statement.

yet it does imply something more might be going on (like reverting values at the end) when in actuality there isn't.

marcoskirsch, 2017-09-20:
There's a performance advantage to the context manager. The advantage will widen once/if we do #188.

All this said, there might be other clever ways to eliminate that performance advantage and we definitely didn't do this for performance reasons. Besides, I will admit it's probably negligible especially to Python users.

We did it because we thought it was nice for DRYness and to visually group configuration of each channel.

epage, 2017-09-20:
Could you expound on why this has a performance advantage and why this helps with #188?

nidaqmx and nixnet support multiple forms of indexing without needing context managers.

epage, 2017-09-20:
> We did it because we thought it was nice for DRYness and to visually group configuration of each channel.

How does this help with DRY.

When I compare
```python
# using a context manager.
with session['2'] as c2:
    c2.voltage_level = 5
    c2.output_enabled = True
```
with
```python
# using a context manager.
c2 = session['2']
c2.voltage_level = 5
c2.output_enabled = True
```

There doesn't seem to be any extra repetition.  The only value I see is the grouping which I imagine is negligible benefit unless you are doing some crazy configurations.

MrElusive, 2017-09-20:
I think the context manager actually adds overhead more than anything.

epage elaborated on my concern [here](https://github.com/ni/nimi-python/issues/187#issuecomment-328920162). I don't think this usage of context management adds any value like it would in init/close or initiate/abort. I wouldn't even call it syntactic sugar, since it actually hurts readability, IMO.

marcoskirsch, 2017-09-22:
> Could you expound on why this has a performance advantage and why this helps with #188?

It would only help vs (a):

    session['2'].voltage_level = 5
    session['2'].output_enabled = True

rather than (b) which is what you wrote:

    c2 = session['2']
    c2.voltage_level = 5
    c2.output_enabled = True

Thinking about it, as with any indexing, customer expectation is that "of course there's a penalty for the indexing". So I am agreeing with you: no concern - we'll teach people to assign to a variable and get rid of the context manager..

> How does this help with DRY?

Would helps with DRY in exhibit (a) but not (b) - because you aren't typing ['2'] over and over.

@epage @MrElusive thanks for the input!

commit 37d3d722 (2017-09-26): Fix #343: _RepeatedCapability should not be a context manager
commit 22b36adc (2017-09-26): Fix #343: _RepeatedCapability should not be a context manager (#352)
closed 2017-09-26 by marcoskirsch