ni/nimi-python #350: config.py references python names for context manager. We should use C names

Fladolcetta, 2017-09-22:
'context_manager_name': {
        'task': 'scan',
        'initiate': 'initiate_scan',

Marcos wrote:

> This refers to a function using its python name.
In all other places we use the C names, for example init_function (see line 24) or when specifying parameters for buffer retrieval mechanisms.
We should be consistent about this and it seems like C names as found in metadata is the way to go.

Fladolcetta, 2017-09-22:
@marcoskirsch 
The value here "initiate_scan", which is what is used in session.py. If we use C names here, than in the session.py.mako we would need to translate the C Name "InitiateScan" to "initiate_scan".

This seems like it would lead to potential issues. Thoughts?

marcoskirsch, 2017-09-25:
> This seems like it would lead to potential issues. Thoughts?

It's how we do it in several other places. We should be consistent.

PR #385 (2017-09-29): Use the actual function name and then lookup python name
[X] This contribution adheres to [CONTRIBUTING.md]
~~[ ] I've updated [CHANGELOG.md](https://github.com/ni/<reponame>/blob/master/CHANGELOG.md) if applicable.~~
### What does this Pull Request accomplish?
* Use C name to lookup context manager actions method names to use
* Rename keys to add '_function' to be clear that it is the function name

### List issues fixed by this Pull Request below, if any.
* Fix #350 

### What testing has been done?
* Travis & System tests
closed 2017-09-29 by marcoskirsch