ni/nimi-python #372: Function metadata should include its name

marcoskirsch, 2017-09-28:
functions.py is organized as a dictionary, where the key is the C function name and the value is its metadata.

This has a couple of drawbacks:

* It's inconsistent with parameters
* It doesn't allow us to control ordering in metadata
* We can't pass a function dictionary alone if we need the function name

This proposes changing functions.py to be a list of dictionaries, each element corresponding to a function. The C name would be one of the key/value pairs in the dictionary.

Fladolcetta, 2017-10-09:
We need to get feedback about the API to understand the importance of the specific order.

marcoskirsch, 2017-10-10:
> We need to get feedback about the API to understand the importance of the specific order.

Meaning: right now Python APIs are using alphabetical ordering. Does method order matter? Should we follow the same order we follow on the C API?

Once we know that, we can change metadata format to a list with a specific order.

Fladolcetta, 2017-11-06:
Once @marcoskirsch did #372 then the only remaining problem is the lack of control over ordering. This is not a concern unless there is a complaint.

closed 2017-11-06 by Fladolcetta