ni/nimi-python #379: Verify 32-bit Python 2 is handling ViInt64 correctly

marcoskirsch, 2017-09-29:
On Python 2, int's size is equals to machine's word size. There's a different type long which can be of any size.
On Python 3, there's no long and int can be of any size.

Our APIs have ViInt64 which is 64-bits. We need to make sure that we are handling values larger than 32-bit correctly in Python 2 32-bit (where word size is 32-bit).

I'm not certain what the best way of doing this is. Possibly we need to inspect Python version and machine bitness and then have class ViInt64 inherit from long rather than int in the Python2 32-bit case.

marcoskirsch, 2017-09-29:
I have added a test for this in PR #383. It needs to be run on 32-bit system to verify there's a problem.

Fladolcetta, 2017-10-17:
Verified!

closed 2017-10-17 by Fladolcetta