ni/nimi-python #381: Python function names are sometimes incorrect

marcoskirsch, 2017-09-29:
I added a function to NI-FAKE: niFake_Use64BitNumber.
This should expand to use_64_bit_number yet it expanded to use64_bit_number.

texasaggie97, 2017-10-10:
What about `niDMM_GetAttributeViInt32`? Should it become `_get_attribute_vi_int_32`? Right now it is `_get_attribute_vi_int32`.

marcoskirsch, 2017-10-10:
Right, same issue.
Which seems right to you?

texasaggie97, 2017-10-10:
It is the same issue, though to me, vi_int_32 sounds funny. Though to be consistent, if we fix the other, we should change this too.

marcoskirsch, 2017-10-10:
Of the following 3:

* _get_attribute_vi_int32
* _get_attribute_viint32
* _get_attribute_vi_int_32

...which do you think is most correct?

marcoskirsch, 2017-11-29:
If all the functions with this problem as of now are private, then we don't *need* to fix this issue.

marcoskirsch, 2017-11-29:
These are all the Session methods with numbers in their names in master as of right now:

```
us-aus-ras-130-164-29-8:nimi-python kirsch$ find * | grep 'generated/.*/session\.py' | xargs grep -ir 'def\ .*[0-9].*(.*)'
generated/niscope/session.py:    def _get_attribute_vi_int32(self, attribute_id):
generated/niscope/session.py:    def _get_attribute_vi_int64(self, attribute_id):
generated/niscope/session.py:    def _get_attribute_vi_real64(self, attribute_id):
generated/niscope/session.py:    def _set_attribute_vi_int32(self, attribute_id, value):
generated/niscope/session.py:    def _set_attribute_vi_int64(self, attribute_id, value):
generated/niscope/session.py:    def _set_attribute_vi_real64(self, attribute_id, value):
generated/nidcpower/session.py:    def _get_attribute_vi_int32(self, attribute_id):
generated/nidcpower/session.py:    def _get_attribute_vi_int64(self, attribute_id):
generated/nidcpower/session.py:    def _get_attribute_vi_real64(self, attribute_id):
generated/nidcpower/session.py:    def _set_attribute_vi_int32(self, attribute_id, attribute_value):
generated/nidcpower/session.py:    def _set_attribute_vi_int64(self, attribute_id, attribute_value):
generated/nidcpower/session.py:    def _set_attribute_vi_real64(self, attribute_id, attribute_value):
generated/nidmm/session.py:    def _get_attribute_vi_int32(self, attribute_id):
generated/nidmm/session.py:    def _get_attribute_vi_real64(self, attribute_id):
generated/nidmm/session.py:    def _set_attribute_vi_int32(self, attribute_id, attribute_value):
generated/nidmm/session.py:    def _set_attribute_vi_real64(self, attribute_id, attribute_value):
generated/nifake/session.py:    def _get_attribute_vi_int32(self, attribute_id):
generated/nifake/session.py:    def _get_attribute_vi_int64(self, attribute_id):
generated/nifake/session.py:    def _get_attribute_vi_real64(self, attribute_id):
generated/nifake/session.py:    def _set_attribute_vi_int32(self, attribute_id, attribute_value):
generated/nifake/session.py:    def _set_attribute_vi_int64(self, attribute_id, attribute_value):
generated/nifake/session.py:    def _set_attribute_vi_real64(self, attribute_id, attribute_value):
generated/nifake/session.py:    def use64_bit_number(self, input):
generated/niswitch/session.py:    def _get_attribute_vi_int32(self, attribute_id):
generated/niswitch/session.py:    def _get_attribute_vi_real64(self, attribute_id):
generated/niswitch/session.py:    def _set_attribute_vi_int32(self, attribute_id, attribute_value):
generated/niswitch/session.py:    def _set_attribute_vi_real64(self, attribute_id, attribute_value):
generated/nifgen/session.py:    def create_waveform_f64(self, waveform_data_array):
generated/nifgen/session.py:    def create_waveform_from_file_f64(self, file_name, byte_order):
generated/nifgen/session.py:    def create_waveform_from_file_i16(self, file_name, byte_order):
generated/nifgen/session.py:    def create_waveform_i16(self, waveform_data_array):
generated/nifgen/session.py:    def _get_attribute_vi_int32(self, attribute_id):
generated/nifgen/session.py:    def _get_attribute_vi_real64(self, attribute_id):
generated/nifgen/session.py:    def _set_attribute_vi_int32(self, attribute_id, attribute_value):
generated/nifgen/session.py:    def _set_attribute_vi_real64(self, attribute_id, attribute_value):
generated/nifgen/session.py:    def write_binary16_waveform(self, waveform_handle, data):
generated/nifgen/session.py:    def write_named_waveform_f64(self, waveform_name, data):
generated/nifgen/session.py:    def write_named_waveform_i16(self, waveform_name, data):
generated/nimodinst/session.py:    def _get_installed_device_attribute_vi_int32(self, index, attribute_id):
```

Of those, the only public ones are:

```
generated/nifake/session.py:    def use64_bit_number(self, input):
generated/nifgen/session.py:    def create_waveform_f64(self, waveform_data_array):
generated/nifgen/session.py:    def create_waveform_from_file_f64(self, file_name, byte_order):
generated/nifgen/session.py:    def create_waveform_from_file_i16(self, file_name, byte_order):
generated/nifgen/session.py:    def create_waveform_i16(self, waveform_data_array):
generated/nifgen/session.py:    def write_binary16_waveform(self, waveform_handle, data):
generated/nifgen/session.py:    def write_named_waveform_f64(self, waveform_name, data):
generated/nifgen/session.py:    def write_named_waveform_i16(self, waveform_name, data):
```

The first one is in NI-FAKE so we don't care.
The rest are all in NI-FGEN. Based on discussion in #511 and #632 we had agreed to remove all flavors of create_waveform and write_waveform that take the waveform as a list of i16. Regardless, the names seem fine to me so I'll just close this.

closed 2017-11-29 by marcoskirsch