ni/nimi-python #404: Multiple issues with revision_query()

injaleea, 2017-10-03:
```
with nidmm.Session('name') as session:
    print(session.revision_query())
```
returns (1,0) for a simulate 4080 and (1,1) for real 408x.
For 4072(both simulated and real) returns following error 
```
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python36-32\lib\sit
e-packages\nidmm\session.py", line 2757, in revision_query
    return python_types.ViChar(instrument_driver_revision_ctype.value), python_t
ypes.ViChar(firmware_revision_ctype.value)
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python36-32\lib\sit
e-packages\nidmm\python_types.py", line 20, in __new__
    return int(val)
ValueError: invalid literal for int() with base 10: b'r'
```

marcoskirsch, 2017-10-03:
From src/nidmm/metadata/functions.py (minus documentation):
```
    'revision_query': {
        'codegen_method': 'public',
        'returns': 'ViStatus',
        'parameters': [
            {
                'direction': 'in',
                'enum': None,
                'name': 'vi',
                'type': 'ViSession',
            },
            {
                'direction': 'out',
                'enum': None,
                'is_buffer': True,
                'name': 'instrumentDriverRevision',
                'type': 'ViChar',
            },
            {
                'direction': 'out',
                'enum': None,
                'is_buffer': True,
                'name': 'firmwareRevision',
                'type': 'ViChar',
            },
        ],
    },
```

The problem is very obvious - the metadata says the type is ViChar.
I think this is another manifestation of https://github.com/ni/nimi-python/issues/396

injaleea, 2017-10-03:
Noticed for 408x devices firmware revisions as number and for 407x and 4065 as revision letters.(From LabVIEW APIs)

```
Device	Instrument Driver Revision	Firware Revision
4065	17.6.0	                                rev F
4070	17.6.0	                                rev M
4080	17.6.0	                               16030313
4081	17.6.0	                               16030313
```

PR #420 (2017-10-06): Clean up string and buffer handling
[X] This contribution adheres to [CONTRIBUTING.md]
~~[ ] I've updated [CHANGELOG.md](https://github.com/ni/<reponame>/blob/master/CHANGELOG.md) if applicable.~~
### What does this Pull Request accomplish?
* All string types are consolidated to ViChar
  * Remove [] from name
  * Change ViString, ViConstString & ViRsrc to ViChar
  * Set 'is_buffer' to true
* Encode/decode page is now determined by variable on Session `self._encoding`
  * Defaults to 'windows-1251'
  * No mechanism to change in Session creation
* Add size information to revistion_query
  * Allo [... cut]
closed 2017-10-06 by marcoskirsch