ni/nimi-python #417: Function enums for nidmm doesn't list not updated for 408x

injaleea, 2017-10-05:
Enums.py for nidmm should be updated with functions supported by 408x.  As of now it lists as follows.

```
class Function(Enum):
    DC_VOLTS = 1
    '''
    All devices supported.
    '''
    AC_VOLTS = 2
    '''
    All devices supported.
    '''
    DC_CURRENT = 3
    '''
    All devices supported.
    '''
    AC_CURRENT = 4
    '''
    All devices supported.
    '''
    _2_WIRE_RESISTANCE = 5
    '''
    All devices supported.
    '''
    _4_WIRE_RESISTANCE = 101
    '''
    NI 4065, and NI 4070/4071/4072 supported.
    '''
    FREQUENCY = 104
    '''
    NI 4070/4071/4072 supported.
    '''
    PERIOD = 105
    '''
    NI 4070/4071/4072 supported.
    '''
    TEMPERATURE = 108
    '''
    NI 4065, and NI 4070/4071/4072 supported.
    '''
    _AC_VOLTS_DC_COUPLED = 1001
    '''
    NI 4070/4071/4072 supported.
    '''
    DIODE = 1002
    '''
    All devices supported.
    '''
    WAVEFORM_VOLTAGE = 1003
    '''
    NI 4070/4071/4072 supported.
    '''
    _WAVEFORM_CURRENT = 1004
    '''
    NI 4070/4071/4072 supported.
    '''
    CAPACITANCE = 1005
    '''
    NI 4072 supported.
    '''
    INDUCTANCE = 1006
    '''
    NI 4072 supported.
    '''
```

texasaggie97, 2017-10-05:
Updated internal CAR to mention enum help as well.

Fladolcetta, 2017-10-09:
Closing this issue as this is being tracked internally for the source of the metadata

closed 2017-10-09 by Fladolcetta