ni/nimi-python #440: Need to create way to remove size on an input array from parameter list.

Fladolcetta, 2017-10-09:
Right now we have a function such as the following:

`session.array_input_function(test_array_size, test_array)`

We really only want to do the following:

`session.array_input_function(test_array)`

Where in functions_addon.py we specify that test_array_size is len(test_array).

marcoskirsch, 2017-10-09:
I think this will require more metadata, something like a new size mechanism:

`['size'] = {'mechanism':'len', 'value':'testArraySize'} `

marcoskirsch, 2017-10-09:
Note: we saw the need for this in niDCPower_SetSequence()

Fladolcetta, 2017-10-10:
Also needed for:

- nifake array_input_function (number_of_elements)

- nidcpower create_advanced_sequence (attribute_id_count)

PR #452 (2017-10-10): Add 'len' mechanism for buffers 
[X] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).
~~[ ] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.~~
### What does this Pull Request accomplish?
* Add 'len' mechanism for buffer length code generation
  * metadata looks like
    ```
        'ArrayInputFunction': { 'parameters': { 2: { 'size': {'mechanism':'len', 'value':'numberOfElements'}, }, }, },
    ```
  * This mechanism will set the size parameter to the length of the buffer paramet [... cut]
closed 2017-10-10 by texasaggie97