ni/nimi-python #444: niDCPower_MeasureMultiple does not have array size input

marcoskirsch, 2017-10-09:
This is the function signature:

```
ViStatus _VI_FUNC niDCPower_MeasureMultiple(
ViSession vi,
ViConstString channelName,
ViReal64 voltageMeasurements[],
ViReal64 currentMeasurements[]);
```

No parameter tells us the size of the arrays voltageMeasurements and currentMeasurements.
How is this used?
How do we make this work in Python? these are output arrays so we need to figure out how large the buffers we allocate need to be.

marcoskirsch, 2017-10-10:
@injaleea commented in duplicate #445:

> Calling nidcpower measure_multiple causes python environment to crash when doing pytest.
> 
> ```
> import nidcpower
> import pytest
> import re
> 
> 
> @pytest.fixture(scope='function')
> def session():
>     with nidcpower.Session('FakeDevice', '', False, 'Simulate=1, DriverSetup=Model:4143; BoardType:PXIe') as simulated_session:
>         yield simulated_session
> 
> 
> def test_measure_multiple(session):
>     session.source_mode = nidcpower.SourceMode.SINGLE_POINT
>     '''session.configure_aperture_time(0, nidcpower.ApertureTimeUnits.SECONDS)'''
>     #session.voltage_level = 1
>     count = 10
>     #session.measure_when = nidcpower.MeasureWhen.ON_DEMAND
>     with session.initiate():
>         voltage_measurements, current_measurements = session.measure_multiple()
>     assert len(voltage_measurements) == 4   # measuremultiple will return a reading for all channel , since 4143 has 4 channel expecting 4 readings 
>     assert len(current_measurements) == 4 
>     assert isinstance(voltage_measurement[1], float)  
>     assert isinstance(current_measurements[1], float) 
> 
> ```
> Also noticed that assertion on test_get_attribute_string fails sometime when it is expecting 4143  session.instrument_model is returning as 4110.[Failure was consistent and disappeared after reboot]

marcoskirsch, 2017-10-17:
> How is this used?

Client needs to allocate arrays large enough for the number of channels specified in the channelName input. So "" means "all the channels in the session, based on when you opened the session". "0-3" means "4 channels". "0, 1" means 2 channels.

* **Option 1**: Don't support niDCPower_MeasureMultiple() in Python. We need to evaluate how bad this would be and what the workarounds are.
    * **Option 1b**: ... until a better alternative exists in NI-DCPower.
* **Option 2**: Implement channel parsing logic in Python so it matches the NI-DCPower driver. This would be error prone, and a shame.
* **Option 3**: Use an internal, secret, private function in NI-DCPower that does the parsing for you. This exists and is leverage by the C# API which has the same problem. But that's a closed source API. Do we want to use a private function in our Open Source Python wrappers? Ideally no.

thofler, 2017-10-18:
> Option 1: Don't support niDCPower_MeasureMultiple() in Python. We need to evaluate how bad this would be and what the workarounds are.

It would be preferably to support MeasureMultiple because it makes it easy to take voltage and current measurements on several channels at the same time.  A workaround could be to use a common measure trigger and fetch instead, but the setup is harder (and not possible on devices that do not support triggering such as 4110 and 4130).

> Option 3: Use an internal, secret, private function in NI-DCPower that does the parsing for you. This exists and is leverage by the C# API which has the same problem. But that's a closed source API. Do we want to use a private function in our Open Source Python wrappers? Ideally no.

Option 3b: Publicly support the function that does the parsing for you in NI-DCPower.

marcoskirsch, 2017-10-18:
> Option 3b: Publicly support the function that does the parsing for you in NI-DCPower.

The timeline for 3b is long. So we'd have to at least temporarily do 3 or take a while until function is supported.

marcoskirsch, 2017-10-24:
**Option 4**: Do the IVI-dance! Change the behavior of the function so that when VI_NULL is passed for voltageMeasurements and currentMeasurements, the size is returned. We can then allocated and call again.

* Pros: No new functions. No exposing private functions. No changing existing behavior.
* Cons: Would require driver work.

commit 2c202f82 (2017-12-18): Add support for measure_multiple

Fixes #444
Clean up other nidcpower system tests to use fixture.
PR #698 (2017-12-18): Add nidcpower.Session.measure_multiple
[X] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

[X] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.

[X] I've added tests applicable for this pull request

### What does this Pull Request accomplish?

* Adds support for `nidcpower.Session.measure_multiple`. This was not trivial because it requires access to an undocumented function.
* Clean up nidcpower system tests to use fixtures whenever possible.

### List issues fixed by this Pul [... cut]
commit eaeb7e58 (2017-12-18): Add nidcpower.Session.measure_multiple (#698)

* Add support for measure_multiple

Fixes #444
Clean up other nidcpower system tests to use fixture.

* Add test for measure_multiple and use fixture if possible

* Update generated

* update with new measure_multiple function
closed 2017-12-18 by marcoskirsch