ni/nimi-python #466: How to use a _RepeatedCapability object for nidcpower ?

injaleea, 2017-10-12:
Running code snippet
```
with nidcpower.Session('4143', '', False, 'Simulate=1, DriverSetup=Model:4143; BoardType:PXIe') as session:
    channel = session['0']
    default_output_function = channel.output_function
    channel.function = nidcpower.OutputFunction.DC_CURRENT
```
retruning Type Error 

> E           TypeError: <nidcpower.session._RepeatedCapability object at 0x0320D3 70> is a frozen class

When setting output function for the second time to DC_CURRENT

marcoskirsch, 2017-10-12:
If the name of the property in incorrect, then it appears like you’re adding a new property to the object. We disallow it, that’s the error you’re getting.

So it is intended behavior. Do you have any suggestions on how to improve the error? Then we can reword this issue and tag it as enhancement/API.

injaleea, 2017-10-13:
Shouldn't be saying object has no attributes when calling an invalid function instead of telling the user about frozen class ?

PR #487 (2017-10-19): Improve returned error for invalid properties.
[x ] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

[ x] I've added any unit tests that are applicable for this pull request

### What does this Pull Request accomplish?

Gives a better error for setting an invalid attribute. This error is identical to the error one might get when getting an invalid attribute.

### List issues fixed by this Pull Request below, if any.
* Fix #466 

### What testing has been done?

Tox and system tests.
closed 2017-10-19 by marcoskirsch