ni/nimi-python #489: Confusing errors when building session.py and functions.rst

Fladolcetta, 2017-10-18:
When I run tox I get the following errors:

> FGEN:
   session.py
      Warning: "init" not found in function metadata. Typo? Generated code will be funky!
   functions.rst
      Warning: "init" not found in function metadata. Typo? Generated code will be funky!
DCPower
   session.py
      Warning: "ClearError" not found in function metadata. Typo? Generated code will be funky!
      Warning: "MeasureMultiple" not found in function metadata. Typo? Generated code will be funky!
      Warning: "ConfigureOutputFunction" not found in function metadata. Typo? Generated code will be funky!
      Warning: "ConfigureCurrentLimit" not found in function metadata. Typo? Generated code will be funky!
      Warning: "ConfigureOutputFunction" not found in function metadata. Typo? Generated code will be funky!
      Warning: "ConfigureVoltageLimit" not found in function metadata. Typo? Generated code will be funky!
      Warning: "ConfigureOutputEnabled" not found in function metadata. Typo? Generated code will be funky!
      Warning: "ConfigureOutputEnabled" not found in function metadata. Typo? Generated code will be funky!
   functions.rst
      Warning: "MeasureMultiple" not found in function metadata. Typo? Generated code will be funky!
      Warning: "ConfigureOutputFunction" not found in function metadata. Typo? Generated code will be funky!
      Warning: "ConfigureCurrentLimit" not found in function metadata. Typo? Generated code will be funky!
      Warning: "ConfigureOutputFunction" not found in function metadata. Typo? Generated code will be funky!
      Warning: "ConfigureVoltageLimit" not found in function metadata. Typo? Generated code will be funky!
DMM
   session.py
      Warning: "ConfigureThermistorType" not found in function metadata. Typo? Generated code will be funky!
      Warning: "ConfigureThermistorType" not found in function metadata. Typo? Generated code will be funky!
      Warning: "ConfigureThermistorType" not found in function metadata. Typo? Generated code will be funky!
   functions.rst
      Warning: "ConfigureThermistorType" not found in function metadata. Typo? Generated code will be funky!
      Warning: "ConfigureThermistorType" not found in function metadata. Typo? Generated code will be funky!
      Warning: "ConfigureThermistorType" not found in function metadata. Typo? Generated code will be funky!
NI-SWITCH
   session.py
      Warning: "ClearError" not found in function metadata. Typo? Generated code will be funky!
      Warning: "InitWithOptions" not found in function metadata. Typo? Generated code will be funky!
      Warning: "init" not found in function metadata. Typo? Generated code will be funky!
      Warning: "init" not found in function metadata. Typo? Generated code will be funky!
      Warning: "InitWithOptions" not found in function metadata. Typo? Generated code will be funky!
   functions.rst
      Warning: "InitWithOptions" not found in function metadata. Typo? Generated code will be funky!
      Warning: "init" not found in function metadata. Typo? Generated code will be funky!

marcoskirsch, 2017-10-18:
https://github.com/ni/nimi-python/pull/479 exposes a lot more of these warnings, this was intentional.

As it stands, documentation for functions that are included in the Python APIs reference functions that are not included in the Python APIs. This means we generate broken links in the RST documentation. These warnings at least help us get the full list and understand it. In reality, we should be stricter and also catch references to private methods.

How to address the problem is not trivial. In some cases we can modify the source documentation. In some cases we may need to do replacements or override it from the Python side.

Fladolcetta, 2017-11-06:
If we can't fix this all the way we should at least stop creating links

marcoskirsch, 2019-11-06:
I think one way to fix (most?) of these is by adding metadata that says:
"For this function, the corresponding property is XXX". Then at codegen we can reference the property rather than the non-existent function.

A better way of fixing it, of course, would be to reword documentation.

marcoskirsch, 2020-01-24:
Since this is non-trivial codegen noise, I wanted to do a bit of analysis to see how many functions are actually referenced in documentation that are not present in metadata - which is what leads to this warning. As I suspected, a lot of the warnings are duplicates.

First I ran full codegen and stored in a file so I can look at it.
```
python3 -m tox -e clean
python3 -m tox -e codegen > codegen.log
```

The total number of "funky" warnings emitted is:
```
sort codegen.log | grep funky | wc -l
     350
```

But indeed there is a lot of duplicates:
```
(base) kirsch@Kirsch0 nimi-python % sort codegen.log | grep funky | uniq -c | sort -nr
  60 Warning: "OutputFunction" not found in function metadata. Typo? Generated code will be funky!
  36 Warning: "OverrangingEnabled" not found in function metadata. Typo? Generated code will be funky!
  36 Warning: "ComplianceLimitSymmetry" not found in function metadata. Typo? Generated code will be funky!
  24 Warning: "PulseVoltageLimitRange" not found in function metadata. Typo? Generated code will be funky!
  24 Warning: "PulseCurrentLimitRange" not found in function metadata. Typo? Generated code will be funky!
  16 Warning: "CreateWaveformFromFileHWS" not found in function metadata. Typo? Generated code will be funky!
  13 Warning: "InitWithOptions" not found in function metadata. Typo? Generated code will be funky!
  12 Warning: "VoltageLimitRange" not found in function metadata. Typo? Generated code will be funky!
  12 Warning: "ExportSignal" not found in function metadata. Typo? Generated code will be funky!
  12 Warning: "CurrentLimitRange" not found in function metadata. Typo? Generated code will be funky!
  11 Warning: "ConfigureOutputMode" not found in function metadata. Typo? Generated code will be funky!
  10 Warning: "ActualNumwfms" not found in function metadata. Typo? Generated code will be funky!
   9 Warning: "Initiate_Scan" not found in function metadata. Typo? Generated code will be funky!
   7 Warning: "init" not found in function metadata. Typo? Generated code will be funky!
   6 Warning: "Init" not found in function metadata. Typo? Generated code will be funky!
   6 Warning: "ConfigureTriggerSource" not found in function metadata. Typo? Generated code will be funky!
   3 Warning: "VoltageLimitLow" not found in function metadata. Typo? Generated code will be funky!
   3 Warning: "VoltageLimitHigh" not found in function metadata. Typo? Generated code will be funky!
   3 Warning: "SendSoftwareTrigger" not found in function metadata. Typo? Generated code will be funky!
   3 Warning: "RouteSignalOut" not found in function metadata. Typo? Generated code will be funky!
   3 Warning: "Reset" not found in function metadata. Typo? Generated code will be funky!
   3 Warning: "PulseVoltageLimitLow" not found in function metadata. Typo? Generated code will be funky!
   3 Warning: "PulseVoltageLimitHigh" not found in function metadata. Typo? Generated code will be funky!
   3 Warning: "PulseCurrentLimitLow" not found in function metadata. Typo? Generated code will be funky!
   3 Warning: "PulseCurrentLimitHigh" not found in function metadata. Typo? Generated code will be funky!
   3 Warning: "PulseBiasVoltageLimitLow" not found in function metadata. Typo? Generated code will be funky!
   3 Warning: "PulseBiasVoltageLimitHigh" not found in function metadata. Typo? Generated code will be funky!
   3 Warning: "PulseBiasCurrentLimitLow" not found in function metadata. Typo? Generated code will be funky!
   3 Warning: "PulseBiasCurrentLimitHigh" not found in function metadata. Typo? Generated code will be funky!
   3 Warning: "CurrentLimitLow" not found in function metadata. Typo? Generated code will be funky!
   3 Warning: "CurrentLimitHigh" not found in function metadata. Typo? Generated code will be funky!
   3 Warning: "ConfigureTriggerMode" not found in function metadata. Typo? Generated code will be funky!
   3 Warning: "ConfigureMeasurement" not found in function metadata. Typo? Generated code will be funky!
   2 Warning: "FetchWaveformMeasurementArray" not found in function metadata. Typo? Generated code will be funky!
   2 Warning: "CalEnd" not found in function metadata. Typo? Generated code will be funky!
   1 Warning: "ClearError" not found in function metadata. Typo? Generated code will be funky!
```

...this is only this many unique warnings:
```
(base) kirsch@Kirsch0 nimi-python % sort codegen.log | grep funky | uniq -c | sort -nr | wc -l
      36
```