ni/nimi-python #511: Should we support numpy in the APIs

texasaggie97, 2017-10-19:
Right now we take or return Python lists from functions where the underlying driver API takes or returns an array. Should we use numpy in our APIs?

|              | List (s) | numpy (s) | LV* (s) |
|-------------:|----------|-----------|--------|
| 1M samples   | 0.2      | 0.04      | 0.04   |
| 10M samples  | 2.2      | 0.4       | 0.41   |
| 100M samples | 23       | 3.8       | 4.1    |

\* LV zeroes out the buffer before calling into the driver

* Should we use numpy at all?
  * Looks like probably yes
* Should the API have methods for both numpy as well as Python list? Or numpy only?
* Do we need to somehow make the numpy dependency optional?
* Do we allocate the numpy buffer or require the user to allocate?
  * Make it optional and allocate if the user doesn't?

marcoskirsch, 2017-10-20:
Consistency with other NI Python driver APIs (primarily https://github.com/ni/nidaqmx-python and https://github.com/ni/nixnet-python) is very important here. As a client, it would be terrible to deal with this in an inconsistent manner just because I happen to have a digitizer and an MIO board.

marcoskirsch, 2017-10-24:
@texasaggie97, @Fladolcetta, myself met with @epage from [nixnet](https://github.com/ni/nixnet-python) and @neilvana from [nidaqmx](https://github.com/ni/nidaqmx-python/) to discuss this topic, and how to best do this.

Conclusions:

* If we use numpy, it will be as an optional dependency.
* Only use numpy arrays for waveforms / samples. Not for sequence steps, filter coefficients, etc.
* For APIs such as NI-SCOPE that have different "fetch" functions, for different types:
    * Expose only the scaled double version, and have it return list
    * Create fetch_into that take in numpy arrays. These will call the right underlying function based on datatype
* For APIs such as NI-FGEN that have different "write" function for different types:
    * Expose only write, then based on parameter type call appropriate underlying function

epage, 2017-10-24:
For reference, here is the xnet issue: https://github.com/ni/nixnet-python/issues/200

epage, 2017-10-24:
Some inline commentary for the conclusion

- If we use numpy, it will be as an optional dependency.
  - *epage*  Our concern is for the developer who doesn't have a pre-compiled numpy for their target
- Only use numpy arrays for waveforms / samples. Not for sequence steps, filter coefficients, etc.
  - *epage* Considerations: is this config time vs inner loop?  What is the likelihood that the data will be used by or come from numpy?
-  For APIs such as NI-SCOPE that have different "fetch" functions, for different types:
  - *epage* The goal is a "ease of use" API and a "high performance" API.
  - *epage* [nidaqmx's readers/writers](https://github.com/ni/nidaqmx-python/blob/master/nidaqmx/tests/test_stream_analog_readers_writers.py#L55) are a nice approach for the high performance case because they decouple policy and attempt to expose python-like stream manipulation functions.  We recognized that they aren't a perfect fit for every API though and just having variations of the functions can be reasonable.
  - Expose only the scaled double version, and have it return list
    - *epage* The ease of use function we mentioned.  Variants that are meant for performance are left off because that is a numpy thing.
  - Create fetch_into that take in numpy arrays. These will call the right underlying function based on datatype
    - *epage* DAQmx has just a `read` function (on its numpy reader) that can both allocate or reuse a buffer.  We (or at least I) felt it simplifies the API to have an `into` function which I've seen precedence for elsewhere.  If someone cares about performance, they probably also care about their buffer management.

Weird, not sure what went wrong with my markdown.

marcoskirsch, 2017-11-18:
I want to note that NI-SCOPE and NI-FGEN APIs both have "binary" versions of their read/fetch and create/write functions. They exist mostly for performance and memory use and their value in the Python API is diminished if we have to copy each point one by one into a Python list. So basically, no point including them in anything other than numpy flavored methods.

Fladolcetta, 2017-12-20:
Closing as this is now fixed.

closed 2017-12-20 by Fladolcetta