ni/nimi-python #514: Should we support complex numbers in our NI-FGEN and NI-SCOPE APIs?

Fladolcetta, 2017-10-20:


marcoskirsch, 2017-10-21:
There are challenges and API design questions on how to best support those types (NIComplexF64 and NIComplexI16) in our Python APIs.

We estimate that only a very small number of clients use complex waveforms with NI-FGEN and NI-SCOPE – most of those customers use NI-RFSG and NI-RFSA which aren't in nimi-python and aren't planned for the [0.9 release](https://github.com/ni/nimi-python/milestone/2).

Given this, and the sheer amount of work remaining, there's a pretty good change that we won't get to this.

Fladolcetta, 2017-10-23:
Closing item until demand is requested for Complex numbers.

marcoskirsch, 2017-11-29:
When we add this to NI-FGEN, we should keep the same signature for the public method and dispatch to the proper function internally if possible, rather than adding new public methods.

MrDuff8, 2019-07-25:
We expect this is a low-priority item, and we haven't heard requests for it. It's very rare for this to be used on its own without an upconverter or downconverter, which we don't offer support for in python. Please comment if this is something your application needs.

closed 2019-07-25 by MrDuff8