ni/nimi-python #531: Support 'ResetAttribute'

marcoskirsch, 2017-10-26:
Some of the APIs support resetting attributes back to their initial state. This is done through function such as:

`ViStatus _VI_FUNC niFgen_ResetAttribute         (ViSession vi, ViConstString channelName, ViAttr attributeId);`
 
We would want to expose this the same way as [nidaqmx](https://github.com/ni/nidaqmx-python) did.

Fladolcetta, 2017-10-26:
This is not a supported feature on all drivers. Deferring for now.

sbethur, 2020-04-01:
> Some of the APIs support resetting attributes back to their initial state

nidigital is one such API, the entry point in C API is:
```
ViStatus _VI_FUNC niDigital_ResetAttribute(
    ViSession vi,
    ViConstString channelName,
    ViAttr attributeId);
```

marcoskirsch, 2022-04-14:
Most driver runtimes that `nimi-python` provides bindings for don't support this feature.
There is also no customer demand for it.
Will close.

closed 2022-04-14 by marcoskirsch