ni/nimi-python #534: Bad NI-FGEN metadata for GetFIRFilterCoefficients

marcoskirsch, 2017-10-27:
1. Parameter coefficientsArray should have "direction":"out"
2. Parameter numberOfCoefficientsRead should have "direction":"out"
3. Parameter numberOfCoefficientsRead should not be a buffer

```
    'GetFIRFilterCoefficients': {
        'codegen_method': 'public',
        'returns': 'ViStatus',
        'parameters': [
            {
                'direction': 'in',
                'enum': None,
                'name': 'vi',
                'type': 'ViSession',
                'documentation': {
                'description': '''
                Identifies your instrument session. **vi** is obtained from the
                niFgen\_init, nifgen\_InitWithOptions, or nifgen\_InitializeWithChannels
                functions and identifies a particular instrument session.
                ''',
                },
            },
            {
                'direction': 'in',
                'enum': None,
                'name': 'channelName',
                'type': 'ViConstString',
                'documentation': {
                'description': '''
                Specifies the channel name for which you want to configure the operation
                mode.

                **Defined Value**: "0"
                ''',
                },
            },
            {
                'direction': 'in',
                'enum': None,
                'name': 'arraySize',
                'type': 'ViInt32',
                'documentation': {
                'description': 'Specifies the size of the coefficient array',
                },
            },
            {
                'direction': 'in',
                'enum': None,
                'name': 'coefficientsArray',
                'type': 'ViReal64[]',
                'documentation': {
                'description': '''
                Specifies the array of data the onboard signal processor uses for the
                FIR filter coefficients. For the NI 5441, provide a symmetric array of
                95 coefficients to this parameter. The array must have at least as many
                elements as the value that you specify in the **numberOfCoefficients**
                parameter in this function.
                The coefficients should range between –1.00 and +1.00.
                ''',
                },
            },
            {
                'direction': 'in',
                'enum': None,
                'name': 'numberOfCoefficientsRead',
                'type': 'ViInt32[]',
                'documentation': {
                'description': '''
                Specifies the array of data containing the number of coefficients you
                want to read.
                ''',
                },
            },
        ],
        'documentation': {
        'description': '''
        | Returns the FIR filter coefficients used by the onboard signal
        processing block. These coefficients are determined by NI-FGEN and
        based on the FIR filter type and corresponding attribute (Alpha,
        Passband, BT) unless you are using the custom filter. If you are using
        a custom filter, the coefficients returned are those set with the
        nifgen\_ConfigureCustomFIRFilterCoefficients function coerced to the
        quantized values used by the device.
        | To use this function, first call an instance of the
        niFgen\_GetFIRFilterCoefficients function with the
        **coefficientsArray** parameter set to VI\_NULL. Calling the function
        in this state returns the current size of the **coefficientsArray** as
        the value of the **numberOfCoefficientsRead** parameter. Create an
        array of this size, and call the niFgen\_GetFIRFilterCoefficients
        function a second time, passing the new array as the
        **coefficientsArray** parameter and the size as the **arraySize**
        parameter. This second function call populates the array with the FIR
        filter coefficients.
        | Refer to the FIR Filter topic for your device in the *NI Signal
        Generators Help* for more information about FIR filter coefficients.
        This function is supported only for the NI 5441.
        | **Default Value**: None
        ''',
        },
    },
```

marcoskirsch, 2017-10-27:
I added a workaround in src/nifgen/metadata/functions_addon.py
Once this is fixed, that workaround should be removed (search for functions_issues)

marcoskirsch, 2017-10-27:
FWIW: I believe this is a problem in the source metadata. The function's public documentation has several errors: http://zone.ni.com/reference/en-XX/help/370524T-01/fgencref/cvinifgen_getfirfiltercoefficients/

This may be another NI internal bug we need to file, once we investigate it and rule out the metadata extraction tools.

Fladolcetta, 2018-06-07:
All,

Closing this issue per #891 

> We no longer care about #534

commit 9abcd713 (2017-10-30): Fix #534
commit 6f94632f (2017-10-30): Produce correct code for functions that return an array whose size is determined via IVI-dance (#542)

* Update generated

* Add GetArrayUsingIVIDance

* Update generated

* Add test_get_array_using_ivi_dance

* Fix #534

* Update generated

* flake8

* Review comments

* remove redundant code from test_get_an_ivi_dance_string
closed 2018-06-07 by Fladolcetta