ni/nimi-python #535: Generated documentation for return parameters is incomplete

marcoskirsch, 2017-10-27:
Example: NI-FGEN function:

```
    def get_fir_filter_coefficients(self):
        '''get_fir_filter_coefficients

        | Returns the FIR filter coefficients used by the onboard signal
          processing block. These coefficients are determined by NI-FGEN and
          based on the FIR filter type and corresponding attribute (Alpha,
          Passband, BT) unless you are using the custom filter. If you are using
          a custom filter, the coefficients returned are those set with the
          nifgen_ConfigureCustomFIRFilterCoefficients function coerced to the
          quantized values used by the device.
        | To use this function, first call an instance of the
          get_fir_filter_coefficients function with the
          **coefficientsArray** parameter set to VI_NULL. Calling the function
          in this state returns the current size of the **coefficientsArray** as
          the value of the **numberOfCoefficientsRead** parameter. Create an
          array of this size, and call the get_fir_filter_coefficients
          function a second time, passing the new array as the
          **coefficientsArray** parameter and the size as the **arraySize**
          parameter. This second function call populates the array with the FIR
          filter coefficients.
        | Refer to the FIR Filter topic for your device in the *NI Signal
          Generators Help* for more information about FIR filter coefficients.
          This function is supported only for the NI 5441.
        | **Default Value**: None

        Tip:
        This method requires repeated capabilities (usually channels). If called directly on the
        nifgen.Session object, then the method will use all repeated capabilities in the session.
        You can specify a subset of repeated capabilities using the Python index notation on an
        nifgen.Session instance, and calling this method on the result.:

            session['0,1'].get_fir_filter_coefficients()

        Args:
            array_size (int): Specifies the size of the coefficient array

        Returns:
            number_of_coefficients_read (int): Specifies the array of data containing the number of coefficients you
                want to read.
        '''
```

It should have an additional "coefficients_array" return value, before number_of_coefficients_read.

texasaggie97, 2017-11-17:
This cannot be fixed until #516 is fixed. Then we will need to decide if this function is required in nifgen.

marcoskirsch, 2018-04-26:
`niscope.Session.get_frequency_response(self)` appears to have the same problem.

texasaggie97, 2018-04-30:
We explicitly filter out 'ivi-dance' types from the return parameter list.

PR #891 (2018-06-08): Disable nifgen get_fir_filter_coefficients()
- [X] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).
- [X] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.
- [X] I've added tests applicable for this pull request
### What does this Pull Request accomplish?
* Remove `get_fir_filter_coefficients()` because it only applies for OSP and OSP is not included in the Python API yet in general
* Keep all created files and metadata in case we need to enable again in the future
  * Don't want  [... cut]
closed 2018-06-08 by marcoskirsch