ni/nimi-python #538: Several function enums missing from NI-FGEN metadata

marcoskirsch, 2017-10-30:
In src/nifgen/metadata/enums.py, several enums needed by nifgen Session methods are missing.

* **ByteOrder** used by **CreateWaveformFromFileF64** and **CreateWaveformFromFileI16**
* **Signal** used by **ExportSignal**
* **RelativeTo** used by **SetNamedWaveformNextWritePosition** and **SetWaveformNextWritePosition**
* **HardwareState** used by **GetHardwareState**
* **Trigger** used by **SendSoftwareEdgeTrigger**

(final names for the enums may be different)

marcoskirsch, 2017-10-31:
@texasaggie97 is looking into changing the internal process used to extract the initial API metadata. This would include enum information and fix this issue.

injaleea, 2017-11-03:
Need to enable **test_get_hardware_state** when this fixed

bhaswath, 2017-11-20:
When adding Triggers enum to the enums_addon.py; making and installing nifgen and running its system tests, I get the following error :

 _______ ERROR collecting src/nifgen/system_tests/test_system_nifgen.py ________
src\nifgen\system_tests\test_system_nifgen.py:1: in <module>
    import nifgen
..\AppData\Local\Programs\Python\Python36-32\lib\site-packages\nifgen\__init__.py:7: in <module>
    from nifgen.session import Session  # noqa: F401
<frozen importlib._bootstrap>:961: in _find_and_load
    ???
<frozen importlib._bootstrap>:950: in _find_and_load_unlocked
    ???
<frozen importlib._bootstrap>:646: in _load_unlocked
    ???
<frozen importlib._bootstrap>:616: in _load_backward_compatible
    ???
..\AppData\Local\Programs\Python\Python36-32\lib\site-packages\_pytest\assertion\rewrite.py:212: in load_module
    py.builtin.exec_(co, mod.__dict__)
..\AppData\Local\Programs\Python\Python36-32\lib\site-packages\nifgen\session.py:2747: in <module>
    class Session(_SessionBase):
..\AppData\Local\Programs\Python\Python36-32\lib\site-packages\nifgen\session.py:2930: in Session
    def configure_digital_edge_script_trigger(self, trigger_id, source, edge=enums.ScriptTriggerDigitalEdgeEdge.RISING):
..\AppData\Local\Programs\Python\Python36-32\lib\enum.py:324: in __getattr__
    raise AttributeError(name) from None
E   AttributeError: RISING
_______ ERROR collecting src/nifgen/system_tests/test_system_nifgen.py ________
src\nifgen\system_tests\test_system_nifgen.py:1: in <module>
    import nifgen
..\AppData\Local\Programs\Python\Python36-32\lib\site-packages\nifgen\__init__.py:7: in <module>
    from nifgen.session import Session  # noqa: F401
<frozen importlib._bootstrap>:961: in _find_and_load
    ???
<frozen importlib._bootstrap>:950: in _find_and_load_unlocked
    ???
<frozen importlib._bootstrap>:646: in _load_unlocked
    ???
<frozen importlib._bootstrap>:616: in _load_backward_compatible
    ???
..\AppData\Local\Programs\Python\Python36-32\lib\site-packages\_pytest\assertion\rewrite.py:212: in load_module
    py.builtin.exec_(co, mod.__dict__)
..\AppData\Local\Programs\Python\Python36-32\lib\site-packages\nifgen\session.py:2747: in <module>
    class Session(_SessionBase):
..\AppData\Local\Programs\Python\Python36-32\lib\site-packages\nifgen\session.py:2930: in Session
    def configure_digital_edge_script_trigger(self, trigger_id, source, edge=enums.ScriptTriggerDigitalEdgeEdge.RISING):
..\AppData\Local\Programs\Python\Python36-32\lib\enum.py:324: in __getattr__
    raise AttributeError(name) from None
E   AttributeError: RISING
!!!!!!!!!!!!!!!!!!! Interrupted: 2 errors during collection !!!!!!!!!!!!!!!!!!!
=========================== 2 error in 1.05 seconds ===========================

marcoskirsch, 2017-11-20:
> When adding Triggers enum to the enums_addon.py;

Is there a branch on GitHub showing your change so we can try to reproduce?

bhaswath, 2017-11-21:
Created #622 with this change

PR #586 (2017-11-22): Updating enums_addon and system tests for nifgen
[X ] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

~~[] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.~~

[X] I've added tests applicable for this pull request

### What does this Pull Request accomplish?
* Adds enums for NI-Fgen
  * ByteOrder
  * HardwareState
  * RelativeTo
  * Signal
  * Trigger
  * TriggerWhen
* Adds system tests for NI-Fgen

### List issues fixed by this Pull Request below, if any.
* Fix #484  
* Fix #53 [... cut]
closed 2017-11-22 by texasaggie97