ni/nimi-python #557: Creating waveformdata for niFgen ?

injaleea, 2017-11-03:
How to create a waveform data using nifgen ? LabVIEW has niFgen Util Create Bin16 Waveform Data VI. Do we have similar functionality in python ?

injaleea, 2017-11-03:
Found a function in example 
```
def create_waveform_data(number_of_samples):
    waveform_data = []
    angle_per_sample = (2 * math.pi) / number_of_samples
    for i in range(number_of_samples):
        waveform_data.append(math.sin(i * angle_per_sample) * math.sin(i * angle_per_sample * 20))
    return waveform_data
```

injaleea, 2017-11-03:
Above function will work fine for Arbwaveform. Any guidline for creating new ARB sequence ?

marcoskirsch, 2017-11-03:
If you're working on system tests, then the contents of the waveform aren't important. The function from the example creates a signal containing to tones, one at frequency f and one at frequency 20f. For the system_test, simply create a list filled with data from -1.0 to 1.0.

For create_arb_sequence(sequence_length, waveform_handles_array, loop_counts_array):

* sequence_length should be derived from waveform_handles_array. I opened issue #560
* You need to create waveforms and put the handles in a list. That's waveform_handles_array parameter.

closed 2017-11-03 by marcoskirsch