ni/nimi-python #558: HardwareState enum missing for nifgen

injaleea, 2017-11-03:
```
    def get_hardware_state(self):
        '''get_hardware_state

        Returns the current hardware state of the device and, if the device is
        in the hardware error state, the current hardware error.

        Note: Hardware states do not necessarily correspond to NI-FGEN states.

        Returns:
            state (enums.HardwareState): Returns the hardware state of the signal generator.

                **Defined Values**
```
As per above code **get_hardware_state** has to return enums.HardwareState enum. There is no such enum in enums.py

marcoskirsch, 2017-11-03:
Duplicate of #538

closed 2017-11-03 by marcoskirsch

injaleea, 2017-11-03:
Need to enable test_get_hardware_state from system_tests once this got fixed