ni/nimi-python #567: json and xml files produced by system tests are in the root directory

marcoskirsch, 2017-11-03:
`tox -e system_tests` creates:

* junit-dmm-system_tests.xml
* junit-fgen-system_tests.xml
* junit-modinst-system_tests.xml
* junit-nidcpower-system_tests.xml
* junit-scope-system_tests.xml
* junit-switch-system_tests.xml

`tox -e  nidmm_system_tests` produces `nidmm_system_test_result.json` and so forth.

make clean doesn't get rid of these files.

At least we have a .gitignore entry for the XML files but not for the json files.

Adding build products to / is sloppy.

I think this should all go inside `bin/`, maybe under `bin/junit/` and `bin/kibana/` _if we can_ (I understand there are clients for this files).

If _we cannot_ then we should at least make sure they are all in .gitignore and cleaned up during make clean.

marcoskirsch, 2017-11-03:
@injaleea I assigned this to you since you added part of this. Can you verify moving the files off of root is even possible (can kibana consume them if moved?)

injaleea, 2017-11-06:
PR #572 should fix system tests part. Reassigning to @texasaggie97 for fixing junitresults

marcoskirsch, 2017-11-06:
Note that the way you wrote the description of #572, when this PR gets merged the issue will be automatically closed. But that PR is only a partial fix for the issue. So you should either

1. Change the wording so that the issue doesn't get automatically closed or
2. Split up the issue, open a new one for the remaining work.

PR #572 (2017-11-07): updated systemtest environment to report results inside bin folder
[X] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

~~[ ] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.~~

~~[ ] I've added any unit tests that are applicable for this pull request~~

### What does this Pull Request accomplish?

Updating tox to report results as <root>\bin\kibana\ni<driver>_systemtests.json

### List issues fixed by this Pull Request below, if any.

* Fix #567 

### What testing has been done [... cut]
closed 2017-11-07 by marcoskirsch