ni/nimi-python #591: Session methods sort by C function name, not by the method name

marcoskirsch, 2017-11-13:
Session methods are sorted alphabetically by C function name, not by its name used in Python API ('python_name' in metadata).

This is generally the same, but we support providing a manually entered "python_name" in functions_addon.py. In some cases, this means that the generated code shows up in the "wrong" place in the metadata.

marcoskirsch, 2017-11-14:
This was caused by overriding python_name for functions. NI-FAKE shows an example of this in #592. However, I thought it was going to be needed in NI-FGEN but turns out it wasn't. So priority is lowered because the bug is not currently exposed in any of the public APIs.

marcoskirsch, 2018-04-27:
We need to look at property order too. Do we want them sorted by python_name? Or do we want to have control over the ordering so that, for example, we can group related properties.

MrDuff8, 2019-08-01:
For the most part users don't care. So we can rule out manual ordering -- not worth the time. We should sort by python_name rather than things showing up mostly sorted, but not quite.