ni/nimi-python #596: Return values of nifgen_GetFIRFilterCoefficients seem to be incorrect

bhaswath, 2017-11-14:
While writing the test given below, no matter what value was given to coeff_array, the value returned by session.get_fir_filter_coefficients() is always ([], 95). This is different from the behavior on LabVIEW and seems incorrect.

def test_fir_filter_coefficients():
    with nifgen.Session('', False, 'Simulate=1, DriverSetup=Model:5441;BoardType:PXI') as session:
        coeff_array = [1, 0, -1]
        session.configure_custom_fir_filter_coefficients(coeff_array)
        session.commit()
        array, size = session.get_fir_filter_coefficients()
        assert size == len(coeff_array)

marcoskirsch, 2017-11-14:
I'm surprised you aren't getting a driver error. On 5441 FIR filter coefficients must be 95 elements and symmetrical. I vaguely remember that the OSP-related attributes aren't looked at unless you enable OSP. Try that.

bhaswath, 2017-11-17:
Even after enabling OSP, I do not get any error when the FIR Filter coefficients aren't 95 elements and symmetrical. (On LabVIEW as well)

Sorry for not updating this earlier.

PR #891 (2018-06-08): Disable nifgen get_fir_filter_coefficients()
- [X] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).
- [X] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.
- [X] I've added tests applicable for this pull request
### What does this Pull Request accomplish?
* Remove `get_fir_filter_coefficients()` because it only applies for OSP and OSP is not included in the Python API yet in general
* Keep all created files and metadata in case we need to enable again in the future
  * Don't want  [... cut]
closed 2018-06-08 by marcoskirsch