ni/nimi-python #629: Need to pass argument for _actual_meas_wfm_size() method call from fetch_array_measurement()

injaleea, 2017-11-29:
fetch_array_measurement() has an call to private method _actual_meas_wfm_size() without any paramter while _actual_meas_wfm_size() requires one parameter.

From generated:
```
     def fetch_array_measurement(self, timeout, array_meas_function):
        if type(array_meas_function) is not enums.ArrayMeasurement:
            raise TypeError('Parameter mode must be of type ' + str(enums.ArrayMeasurement))
        vi_ctype = visatype.ViSession(self._vi)  # case 1
        channel_list_ctype = ctypes.create_string_buffer(self._repeated_capability.encode(self._encoding))  # case 2
        timeout_ctype = visatype.ViReal64(timeout)  # case 9
        array_meas_function_ctype = visatype.ViInt32(array_meas_function.value)  # case 10
        meas_wfm_size_ctype = visatype.ViInt32(self._actual_meas_wfm_size())  # case 0.0
        meas_wfm_ctype = (visatype.ViReal64 * (self._actual_meas_wfm_size() * self._actual_num_wfms()))()  # case 0.2
        wfm_info_ctype = (waveform_info.struct_niScope_wfmInfo * self._actual_num_wfms())()  # case 0.2
        error_code = self._library.niScope_FetchArrayMeasurement(vi_ctype, channel_list_ctype, timeout_ctype, array_meas_function_ctype, meas_wfm_size_ctype, meas_wfm_ctype, wfm_info_ctype)
        errors.handle_error(self, error_code, ignore_warnings=False, is_error_handling=False)
        return [float(meas_wfm_ctype[i]) for i in range((self._actual_meas_wfm_size() * self._actual_num_wfms()))], [waveform_info.WaveformInfo(wfm_info_ctype[i]) for i in range(self._actual_num_wfms())]
```
```
    def _actual_meas_wfm_size(self, array_meas_function):
        if type(array_meas_function) is not enums.ArrayMeasurement:
            raise TypeError('Parameter mode must be of type ' + str(enums.ArrayMeasurement))
        vi_ctype = visatype.ViSession(self._vi)  # case 1
        array_meas_function_ctype = visatype.ViInt32(array_meas_function.value)  # case 10
        meas_waveform_size_ctype = visatype.ViInt32()  # case 14
        error_code = self._library.niScope_ActualMeasWfmSize(vi_ctype, array_meas_function_ctype, ctypes.pointer(meas_waveform_size_ctype))
        errors.handle_error(self, error_code, ignore_warnings=False, is_error_handling=False)
        return int(meas_waveform_size_ctype.value)
```

marcoskirsch, 2017-11-29:
I'm guessing this is in NI-SCOPE? The issue doesn't say it nor is it tagged.

marcoskirsch, 2017-11-30:
While working on this fix I ran into a problem.

The underlying NI-SCOPE function:
```
ViStatus _VI_FUNC niScope_FetchArrayMeasurement (
   ViSession vi,
   ViConstString channelList,
   ViReal64 timeout,
   ViInt32 arrayMeasFunction,
   ViInt32 measWfmSize,
   ViReal64* measWfm,
   struct niScope_wfmInfo* measWfmInfo);
```
... must be called with a channelList that is not empty or you get the error:

```
           Error: -1074135008: IVI:  (Hex 0xBFFA0020) Unknown channel or repeated capability name.

           Channel name is empty.
```
This is true for other NI-SCOPE functions.

This means that in Python, the client must use a _RepeatedCapability object, which you get by doing something like:

    my_session["0"].fetch_array_measurement(-1, niscope.ArrayMeasurement.ARRAY_GAIN)

So now, inside the implementation of _RepeatedCapability.fetch_array_measurement() we try to call `self._actual_meas_wfm_size()` which is not a member of _RepeatedCapability because it is not a channel-based method - so it only lives in the subclass Session.

I'm open to suggestions on how to fix this.

marcoskirsch, 2017-11-30:
> I'm open to suggestions on how to fix this.

Found a way to do it. PR soon.

commit ea262a1e (2017-11-30): Fix #629 by passing the right arguments to _actual_meas_wfm_size which was moved to _SessionBase
PR #642 (2017-12-01): Fix niscope.Session.active_channel.fetch_array_measurement()
[X] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

~~[ ] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.~~

[X] I've added tests applicable for this pull request

### What does this Pull Request accomplish?

*Allows customization of where a method is generated (class _SessionBase or Session).
    * This used to be deduced from function metadata values for 'has_repeated_capability' and 'is_error_handling'
    * Done  [... cut]
commit af4b373a (2017-12-01): Fix niscope.Session.active_channel.fetch_array_measurement() (#642)

* Add new function metadata 'render_in_session_base'

This tells the generator whether to render the method in _SessionBase
or Session. We used to derive this from ‘has_repeated_capability’ and
‘is_error_handling’ but ran into the need to override the regular
assumption in #629

* Fix #629 by passing the right arguments to _actual_meas_wfm_size which was moved to _SessionBase

* System tests

* Update generated
closed 2017-12-01 by marcoskirsch