ni/nimi-python #630: What need to be passed into ConfigureTriggerDigital ? TRIGGER_SOURCE name or value ?

injaleea, 2017-11-29:
https://github.com/ni/nimi-python/blob/5755ed984aa875f19b57f771c3c94173884484bd/generated/niscope/session.py#L2828

What should be passed to ConfigureTriggerDigital method  as Trigger source ? Can we pass any string to it ? LabVIEW has a Ring control to handle it . So should we be passing name or value of TRIGGER_SOURCE to this ConfigureTriggerDigital ?

Generated code points to Refer to TRIGGER_SOURCE which is removed from enums_addon.py. Do we need to update this ?

Fladolcetta, 2017-11-29:
There should not be an enum for trigger source since it is a string. We do that so customers can use string based routing to route from one device to another. You should use the value from that string enum in labview, or use the strings listed in the niscope.h file.

closed 2017-11-29 by Fladolcetta

marcoskirsch, 2017-11-29:
See https://github.com/ni/nimi-python/wiki/API-design-guidelines, specifically "Encourage use of cross-device trigger routing..." part.