ni/nimi-python #635: Functions that return lists also return the length of the list without reason

epage, 2017-11-29:
```python
    voltage_measurements, current_measurements, in_compliance, actual_count = session.fetch_multiple(session.fetch_backlog)
```
from [nimi-python/nidcpower_measure_record.py at master · ni/nimi-python · GitHub ](https://github.com/ni/nimi-python/blob/master/src/nidcpower/examples/nidcpower_advanced_sequence.py#L39)

 If actual_count is the length of the other returned lists, why is it being returned?  This feels like a vestige of a C API for saying how much of a buffer was actually filled.

Note: other APIs seem to be subject to this, see https://github.com/ni/nimi-python/blob/master/src/nidmm/examples/nidmm_fetch_waveform.py#L25

epage, 2017-11-30:
This is tagged "priority-low".  Is this expressing urgency or importance?

If
- Urgency: the downside is that this will be a source breaker on some of the most commonly called API functions and should be addressed sooner rather than later to minimize impact
- Importance: I'd contend that assessment.

Fladolcetta, 2017-11-30:
Urgency. We have already tagged this as source-breaker.
This might be something we fix before the official 1.0 release.

epage, 2017-11-30:
> This might be something we fix before the official 1.0 release.

FYI since we're following semver, fixing this after 1.0 will mean we bump major and it will be a 2.0 release.

Also, due to the impact on the number of users, this seems bad to defer past 1.0.

texasaggie97, 2017-12-19:
This value is not the size of the array returned. It is how many points in the ~~array~~list are valid.

For example, if you request 1,000,000 points, the array size passed into the C function will be set to 1M samples. We will copy that many in to the list before returning. However, say it timed out, there may only be 420,042 points in the array. That is what `actual_number_of_points` means.

marcoskirsch, 2017-12-19:
This is true for C / LabVIEW. But in Python we raise, so it's not like you can end up with a partially filled list.

texasaggie97, 2017-12-19:
This is also true for the `_into` variants.

Though because we raise, there is no way for them to know how much of it is good.

marcoskirsch, 2017-12-19:
> Though because we raise, there is no way for them to know how much of it is good.

Yeah, if you use the _into variants and an exception occurs, you're out of luck too, I think. There are ways to work around this limitation but they'd be ugly API-wise and probably not worth it.

epage, 2017-12-19:
> This value is not the size of the array returned. It is how many points in the arraylist are valid.

To be clear, I wrote it from the perspective of a python developer, not a C developer.

What does DAQ and XNET do in the underflow case?  I was thinking DAQ might produce a warning instead of raising an exception.

PR #702 (2017-12-20): There are some return parameters that don't make sense in Python so don't return them
[X] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).
[X] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.
[X] I've ~~added~~updated tests applicable for this pull request
### What does this Pull Request accomplish?
* Add option to code generator to allow skipping some parameters in the Python API.
  * Right now, only used for return list

### List issues fixed by this Pull Request below, if any. [... cut]
closed 2017-12-20 by texasaggie97