ni/nimi-python #639: configure_equalization_filter_coefficients fails for niscope

injaleea, 2017-11-30:
calling configure_equalization_filter_coefficients throws  **niscope.errors.Error: -214202: Error description not found** for scope.
```
import niscope
with niscope.Session('FakeDevice', False, True, 'Simulate=1, DriverSetup=Model:5142; BoardType:PXIe') as session:
        fir_conf = [1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
        session.configure_equalization_filter_coefficients(fir_conf)
```
Could set the same with niscope APIs with LabVIEW.

Fladolcetta, 2017-11-30:
What does the IO Trace look like between the two (in labview vs in python)?

jayj2510, 2017-11-30:
> with niscope.Session('FakeDevice', False, True, 'Simulate=1, DriverSetup=Model:5142; BoardType:PXIe') as session:

Not sure if this is the source of your problem, but there is no PXIe-5142.  Only a PXI/PCI.  I'm surprised that it's possible to simulate one.  Maybe the code for this call is messed up for that bus-type.

marcoskirsch, 2017-11-30:
> Not sure if this is the source of your problem, but there is no PXIe-5142.

Fixed in:
https://github.com/ni/nimi-python/pull/643/commits/50d642bcc4bacd9f8d43684d3bceb00dcd6ce23d

But I haven't reenabled the test. I don't rule out that the issue is related, but do note that the exception is thrown after the session is created, so that's weird from NI-SCOPE's part.

@injaleea: If NI-SCOPE allows opening a simulated session to a non-existent model (PXIe-5142 in this case) you should file an internal bug report.

injaleea, 2017-12-01:
> If NI-SCOPE allows opening a simulated session to a non-existent model (PXIe-5142 in this case) you should file an internal bug report.

Similar issue exists with NI-FGen aswell . Could simulate a 5433 with option string '**Simulate=1, DriverSetup=Model:5133; BoardType:PCI**'

Filed internal bug report for both NI-SCOPE and NI-FGen.

injaleea, 2017-12-01:
> What does the IO Trace look like between the two (in labview vs in python)?

From LabVIEW 
```
2.  niScope_ConfigureEqualizationFilterCoefficients (0x0000003F, "0", 35 (0x23), 1.000000 (1.000000E+00))
Process ID: 0x000014E0         Thread ID: 0x0000186C
Start Time: 04:35:35.7572      Call Duration 00:00:00.0000
Status: 0 (VI_SUCCESS)

```
From python
```
> 2.  niScope_ConfigureEqualizationFilterCoefficients (0x00000001, "", 35 (0x23), 1.000000 (1.000000E+00))
> Process ID: 0x00000A90         Thread ID: 0x0000050C
> Start Time: 04:39:15.6922      Call Duration 00:00:00.0750
> Status: 0xFFFCBB46

```
If I am right only thing missmatching is **channel name** . calling ConfigureEqualizationFilterCoefficients as channel specific will fix this issue. 
```
import niscope
with niscope.Session('FakeDevice', False, True, 'Simulate=1, DriverSetup=Model:5142; BoardType:PXIe') as session:
        fir_conf = [1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
        session['0'].configure_equalization_filter_coefficients(fir_conf)
```

marcoskirsch, 2017-12-01:
> If I am right only thing missmatching is channel name . calling ConfigureEqualizationFilterCoefficients as channel specific will fix this issue.

This may be a driver issue, or may be intended behavior. I will ask.

marcoskirsch, 2017-12-01:
Regardless, you can update the system test to pass "0" and we can move on. If the expectation is that the function should work with channel "", then we can reference the NI internal bug (once we have it) in a TODO comment.

marcoskirsch, 2017-12-01:
@injaleea 

Confirmed with NI-SCOPE team member that function is not supposed to work with "". Please file an internal bug report on the issue that the error message is not helpful and outside of the valid IVI error range. Then fix the system test to use "0".

Once this is done, you may close the issue.

closed 2017-12-01 by marcoskirsch