ni/nimi-python #654: niswitch_get_device_info example uses io_resource_descriptor attribute that doesn't exist

marshallgallatin, 2017-12-06:
I tried the niswitch_get_device.py example with a PXI-2575, if you use the -d flag it errors:
```
(venv) C:\Users\Administrator\Desktop\niswitch_python_testing>python niswitch_get_device_info.py -n 2575 -d
Device Info:
Traceback (most recent call last):
  File "niswitch_get_device_info.py", line 22, in <module>
    print(row_format.format('Device Name: ', session.io_resource_descriptor))
AttributeError: 'Session' object has no attribute 'io_resource_descriptor'
```
According to [the documentation](http://nimi-python.readthedocs.io/en/master/niswitch/session.html), the io_resource_descriptor attribute doesn't exist.

I'd fix it but I didn't see an appropriate replacement - logical_name isn't the same thing.

python version:
2.7.13 (v2.7.13:a06454b1afa1, Dec 17 2016, 20:53:40) [MSC v.1500 64 bit (AMD64)]

python -m pip list
enum34 (1.1.6)
niswitch (0.5.0)
pip (9.0.1)
setuptools (38.2.4)
wheel (0.30.0)

marcoskirsch, 2017-12-06:
Why we need #563

marshallgallatin, 2017-12-06:
Also specific_driver_revision is in the example but not on the session too

Fladolcetta, 2017-12-06:
All,

These appears to be removed when @texasaggie97 added attributes_addon. The claim is that these attributes are obsolete, but they are not marked obsolete anywhere else.

PR #673 (2017-12-11): Add back incorrectly removed attributes
[x ] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

### What does this Pull Request accomplish?

Adds back attributes that were incorrectly marked as obsolete. (I used LabVIEW property nodes to verify whether an attribute was obsoleted or not.)
For DMM, I also had to change the enum that was needed for one of the attributes to properly show the name of the enum.

### List issues fixed by this Pull Request below, if any.

* Fix #654 
* Fix #662 

### What testing has been done?

Ran TOX
closed 2017-12-11 by marcoskirsch