ni/nimi-python #656: Provide object oriented approach for conceptual concepts in our APIs

Fladolcetta, 2017-12-06:
For example:

Things like sequences, scripts, waveforms, frequency list, etc should be exposed as types rather than methods on the session.

```
sequence = session.create_advanced_sequence
sequence[0].voltage_level = 1.0
sequence[1].voltage_level = 2.0
```

marcoskirsch, 2017-12-06:
This also may allow us to condense "named" and "handle" based waveform APIs in nifgen. @epage noted this.

Fladolcetta, 2017-12-06:
We can also add this for channels and relays in niswitch. This has less value than other functions.

```
relays['r0c0'].position
```

marcoskirsch, 2017-12-06:
Another place where we can use a little class is for Option String (passed to init) because it's terrible to have to learn some obscure IVI syntax.

marcoskirsch, 2018-04-26:
We've partially done this and planned for a few more instances (i.e. NI-DCPower advanced sequences, NI-SCOPE fancy fetch, etc.)

We will close this so it doesn't distract.

closed 2018-04-26 by marcoskirsch