ni/nimi-python #664: Allow lists of strings for specifying trigger destinations

marcoskirsch, 2017-12-06:
This applies to routing attributes/functions.

For example exported_ref_trigger_output_terminal. This attribute allows the user to specify where they want the exported reference trigger to go to. The driver supports exporting events to multiple destinations simultaneously, and internally this attribute is represented as a list of strings. But, our API represents this attribute as a string containing a comma-separated list of terminals, and we parse that string in the driver to end up with the actual list.
 
I think it'd be nicer if these attributes were instead represented as lists of strings in the Python API.

Fladolcetta, 2017-12-06:
This could apply to many areas, channel names, etc. This would then translate the list to a comma delimited string,

MrDuff8, 2019-08-01:
> This could apply to many areas, channel names, etc.

This already works for channel names or any other repeated capability.

MrDuff8, 2019-08-01:
While this is a valid feature request, we think that the use case of exporting an event to multiple terminals is very rarely used, so we're closing this issue. The workaround is to simply join your list of terminals with a comma. 

``` python 
",".join(my_terminals)
```

closed 2019-08-01 by MrDuff8