ni/nimi-python #740: missing DLL library functions within library.py

ihouses, 2018-02-14:
I want to use the nipower library but I see that is not even possible to call niDCPower_ConfigureOutputEnabled and many other DLL functions since are not defined in the library.py file. It is written that this file is auto generated.

How I suppose to enable/disable output channels, set the current limit and similar with using this library?

marcoskirsch, 2018-02-14:
Hello,

All that functionality is available via properties. See [Session documentation](http://nimi-python.readthedocs.io/en/master/nidcpower/session.html) for a full list, or see [detailed property documentation here](http://nimi-python.readthedocs.io/en/master/nidcpower/attributes.html for detailed documentation).

If you see encounter any problems or have any other input, please provide more feedback. Good luck!

closed 2018-02-14 by marcoskirsch

ihouses, 2018-02-14:
Hello, thanks for the reply.

I am just trying something very basic:

import nidcpower
from nidcpower import enums

channels = '0'
name = 'SMU1'  # PXIe-4139 SMU
session = nidcpower.Session(resource_name=name, channels=channels)

session['0,'].output_enabled = True
session['0,'].voltage_level = 3.1
session['0,'].output_connected = True
session['0,'].output_function = enums.OutputFunction.DC_VOLTAGE
session.commit()

This, somehow changes the output voltage (because if I open the panel I see that) but doesn't enable the output.

marcoskirsch, 2018-02-14:
That's strange. If you don't mind, run NI IO Trace and capture the calls to the library, then paste here?

FYI: #718 changes how channel based properties are accesses. This will be released soon.
`session['0,'].output_enabled = True` becomes `session.channels[0].output_enabled = True`

marcoskirsch, 2018-02-14:
Call `session.initate()` instead of or after `session.commit()` to get the output started.

ihouses, 2018-02-15:
Hello,

I noticed that if session.commit() is not part of a WITH statement nothing happens in the NI trace I/O. That is why the output was always OFF. If I put it within a WITH statement like in the examples then it works fine.

On the other hand I would need to execute commit() without being in the WITH statement.
Is it possible()

marcoskirsch, 2018-02-15:
Are you referring to a `with` statement for the Session creation, or for `initiate()`?
If you post code snippets it will be easier to help. Thanks!

ihouses, 2018-02-16:
Hi,

If you execute: session._initiate() it works

But if you execute session.initiate() doesnt do anything, unless it is after a with statement like:

with session.initialte():
   Do something

But in my case I dont want to use the with statement.

marcoskirsch, 2018-02-16:
That's interesting. Can you post the full code that should but doesn't work?

MrElusive, 2018-02-16:
@marcoskirsch 

I just ran into the same issue:

This code works:
```
with nidcpower.Session("4145") as session:
    with session.initiate():
        voltage_measurement = session['0'].measure(nidcpower.MeasurementTypes.VOLTAGE)
        print("Measurement: {}".format(voltage_measurement))
```

This code doesn't (error -1074118650 (session not running) at the call to measure)
```
with nidcpower.Session("4145") as session:
    session.initiate()
    voltage_measurement = session['0'].measure(nidcpower.MeasurementTypes.VOLTAGE)
    print("Measurement: {}".format(voltage_measurement))
```

marcoskirsch, 2018-02-16:
> I just ran into the same issue

Interesting. Please open an issue about this, as to not hijack this one.