ni/nimi-python #743: nimi-python is not supported in IronPython

MrElusive, 2018-02-16:
Attempting to use the nidcpower module in IronPython results in the following error:
`nidcpower.errors.UnsupportedConfigurationError: System configuration is unsupported: 64bit cli`

This error happens because the IronPython implementation returns "cli" for platform.system(). I assume the other drivers will have the same problem since they all use the same library logic.

A simple workaround is to override the platform.system method:
`
import platform
def get_system():
    return 'Windows' # or 'Linux'
platform.system = get_system
`

marcoskirsch, 2018-02-16:
IronPython is not one of the supported interpreters per NI's support policy, which is:

> * Language versions:
>    * 2.7
>     * 3.4 and later
> * Interpreters:
>     * CPython
>     * PyPy (Latest interpreter version)

Given this, I'm leaning to close as "wontfix" unless there's some justification to keep open.
Thanks!

MrElusive, 2018-02-16:
No problem! I expected it wouldn't be supported. I mostly opened the issue for posterity's sake and to document a workaround.

closed 2018-02-16 by marcoskirsch