ni/nimi-python #744: datetime.timedelta not precise enough for all nimi-python use cases

marcoskirsch, 2018-02-17:
#659 is merged which is great because it's slick. However, it introduced one problem: Python's built-in datetime.timedelta is not precise enough for all of nimi-python's use cases.

For example:
The NI-SCOPE attribute NISCOPE_ATTR_END_OF_RECORD_TO_ADVANCE_TRIGGER_HOLDOFF is specified as a ViReal64 (which is a double) in the C API and can have values with resolution greater than a microsecond.
But in the Python API this corresponds to [`niscope.Session.end_of_record_to_advance_trigger_holdoff`](http://nimi-python.readthedocs.io/en/master/niscope/attributes.html#niscope.Session.end_of_record_to_advance_trigger_holdoff) which is a [datetime.datetime](https://docs.python.org/2/library/datetime.html#timedelta-objects) and can only be specified in resolution of microseconds but not smaller.

marcoskirsch, 2018-02-17:
So how to solve this?

Using float in the Python APIs like before #659 would be an un-pythonic step back.

What would be really neat, is a drop-in, duck-type replacement for datetime.datetime that supports resolutions higher than than the built-in class. Then nimi-python could switch to using that wherever appropriate.

Fladolcetta, 2018-04-12:
The workaround is to use a float in seconds via #796

ni-jfitzger, 2020-05-11:
Additional examples of this problem for niscope:
* Time Per Record: https://nimi-python.readthedocs.io/en/master/niscope/class.html#horz-time-per-record
* Runt Time Low Limit: https://nimi-python.readthedocs.io/en/master/niscope/class.html#runt-time-low-limit
* Runt Time High Limit: https://nimi-python.readthedocs.io/en/master/niscope/class.html#runt-time-high-limit

There may be others; I haven't gone over the API and devices with a fine-toothed comb.
Of these, Time Per Record is the most important and most common, by a long shot.  It is supported by all devices and is commonly used for horizontal configuration, though NI examples tend to steer users toward alternative settings.  It's not unreasonable that a user would configure a scope in such a way that Time Per Record would be less than 1 us.

PR #1449 (2020-05-19): Replace usage of `datetime` library types with `hightime` types
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

- [x] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.

- [x] I've added tests applicable for this pull request

### What does this Pull Request accomplish?

In all nimi-python modules, change the type of applicable properties and method parameters from `datetime.timedelta` to `hightime.timedelta` and from `datetime.datetime` to `hightime.datetime`.

`hightime`  [... cut]
closed 2020-05-19 by marcoskirsch