ni/nimi-python #748: nidcpower.Session.fetch_multiple() API is not Pythonic

marcoskirsch, 2018-02-20:
See [documentation](http://nimi-python.readthedocs.io/en/0.6.0/nidcpower/functions.html#nidcpower.fetch_multiple)

> Return type: | tuple (voltage_measurements, current_measurements, in_compliance, actual_count)
> WHERE
> voltage_measurements (list of float):Returns an array of voltage measurements. Ensure that sufficient space has been allocated for the returned array.
> current_measurements (list of float):Returns an array of current measurements. Ensure that sufficient space has been allocated for the returned array.
> in_compliance (list of bool):Returns an array of Boolean values indicating whether the output was in compliance at the time the measurement was taken. Ensure that sufficient space has been allocated for the returned array.
> actual_count (int):Indicates the number of measured values actually retrieved from the device.

Instead of returning 3 lists and an int, this should return a single list of dict, each with a voltage, current and in_compliance (exact names TBD).

marcoskirsch, 2018-02-20:
There may be other functions in nidcpower that should get similar treatment.
Note this is different than the LabVIEW API.
Note that actual_count should not be returned. This was fixed for other places in #702

MrElusive, 2018-02-20:
"pythonic" aside, I think it's just logical for a measurement to exist as a single object. A namedtuple is a perfect fit for that and more appropriate than a dict.

If a client only wants voltages, it's easy for them to get it: 
```python 
voltage_measurements = map(lambda measurement: measurement.voltage, measurements)
```
Now that looks pythonic :)

thejcannon, 2018-02-20:
@MrElusive beat me too it. I think returning one list is nicer, however a dict with string keys is too much overhead. A named tuple would be the way to go!
I think the current solution is OK, the proposed one (with named tuple) is better 😄

PR #767 (2018-03-15): DCPower Fancy Fetch Multiple
- [X] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).
- [X] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.
- [X] I've ~~added~~updated tests applicable for this pull request
### What does this Pull Request accomplish?
* Add ease of use wrapper for fetch_multiple

### List issues fixed by this Pull Request below, if any.
* Fix #748 

### What testing has been done?
* Unit
* System
closed 2018-03-15 by marcoskirsch