ni/nimi-python #750: General question about multichannel SMUs

ihouses, 2018-02-21:
Hello,

I would like to understand the best way to treat the multichannel SMUs.

In this case when the initiate the SMU: nidcpower.Session(resource_name=name, channels=channels)

Each channel can be treated as a different session?

ihouses, 2018-02-21:
Ok, I fix the problem selecting MANUALLY using the device manager what driver is the correct. (windows proposes a list and you need to select Virtual com port->ST...etc

marcoskirsch, 2018-02-21:
```
In this case when the initiate the SMU: nidcpower.Session(resource_name=name, channels=channels)
Each channel can be treated as a different session?
```

On NI-DCPower devices, you can create one `nidcpower.Session` per each channel, or a single `nidcpower.Session` for all the channels, or anything in between (like one `nidcpower.Session` for channel 0 and 1, another for channel 2 and 3). It's up to you. It depends on your application. Do you want to initiate all channels at once? Or do you need independent state control over them?

> Ok, I fix the problem selecting MANUALLY using the device manager what driver is the correct. (windows proposes a list and you need to select Virtual com port->ST...etc

I assume you posted this in the wrong place.

ihouses, 2018-02-24:
Thanks for your answer. I have a 4 channels SMU and for me the most convenient way is to treat each channel as a single SMU. The code is making this very easy by instantiating a different session for each channel.

Sorry for my previous post as you say I posted it in a wrong place :)

closed 2018-02-25 by marcoskirsch