ni/nimi-python #753: Is data streaming using fetch function possbile for single record continuous data acquisition?

swapnil15689, 2018-03-01:
Basically, I refer to the below example using labview VI for NIscope.

https://forums.ni.com/t5/Example-Program-Drafts/niScope-Single-Channel-Stream-to-Disk/ta-p/3522594

And I wish to do the same using niscope python module.
To be specific, I want to fetch a single record continuously for a very long time. And to make this possible I must write data to a file while acquisition is in process. 
I can do everything in this example using python niscope module except writing the data simultaneously to the binary file. So, at the end I am limited with the available memory.

Currently is niscope module adapted to do this? I do not see any similar method in niscope module which might do the same.

texasaggie97, 2018-03-01:
The niscope module does not provide the binary fetch methods directly. Instead, you would use the [fetch_into](http://nimi-python.readthedocs.io/en/0.7.0/niscope/functions.html#niscope.Session.fetch_into) method. 

Basically, what you would do is create a `numpy.ndarray` with the correct data type and then pass that to `fetch_info()`. This will fill in the array with samples based on the size.

This is based on 0.7.0 (what us currently on PyPI)
``` python
import niscope
import numpy

with niscope.Session('dev1') as session:
    # configure the session as desired
    num_records = 2
    channel_list = [0,1]
    num_channels = len(channel_list)
    record_length = 1000
    voltage_range = 10.0
    session.configure_vertical(voltage_range , niscope.VerticalCoupling.AC)
    session.configure_horizontal_timing(50000000, record_length, 50.0, num_records, True)
    wfm = numpy.ndarray(num_channels * record_length * num_records, dtype=numpy.int16)
    wfm_info = session.channel[channel_list]fetch_into(wfm=wfm)
    record0channel0 = wfm[0:record_length]
    record0channel1 = wfm[record_length:2 * record_length - 1]
    record1channel0 = wfm[2 * record_length:3 * record_length - 1]
    record1channel1 = wfm[3 * record_length:4 * record_length - 1]
```

You would then need to parse through the wfm to get the individual records such that

* index 0 = record 0, channel 0
* index *record_length* = record 0, channel 1
* index 2 \* *record_length* = record 1, channel 0
* index 3 \* *record_length* = record 1, channel 1

Starting with 0.8.0 (under active development) the per record waveform will be added to the wfm_info when using Python 3 (Python 2 is missing some functionality we need to do this)

Here is what the same functionality will look like with 0.8.0 when using Python 3:
``` python
import niscope
import numpy

with niscope.Session('dev1') as session:
    # configure the session as desired
    num_records = 2
    channel_list = [0,1]
    num_channels = len(channel_list)
    record_length = 1000
    voltage_range = 10.0
    session.configure_vertical(voltage_range , niscope.VerticalCoupling.AC)
    session.configure_horizontal_timing(50000000, record_length, 50.0, num_records, True)
    wfm = numpy.ndarray(num_channels * record_length * num_records, dtype=numpy.int16)
    wfm_info = session.channel[channel_list]fetch_into(wfm=wfm, num_records=num_records)
    record0channel0 = wfm_info[0].wfm
    record0channel1 = wfm_info[1].wfm
    record1channel0 = wfm_info[2].wfm
    record1channel1 = wfm_info[3].wfm
```

As mentioned, this portion of the niscope API is being actively developed so the actual usage may change more by the time 0.8.0 is release.

There isn't much documentation on this yet, though it is being worked on. If you want to see the documentation based on what will become the 0.8.0 release, it is available on [readthedocs](http://nimi-python.readthedocs.io/en/docs-fancy_fetch/niscope.html).

One other issue you may run into is not related to niscope but because of the language. LabVIEW is inherently multi threaded, so you can use implement a producer/consumer loop very easily. In Python you will need to handle you own threading and any synchronization required.

swapnil15689, 2018-03-02:
I think for the moment function "fetch_into" needs a slight modification.

As per the source code, we need to pass the information of 'num_samples' to this function.

It needs to be modified from===>
 def fetch_into(self, wfm, timeout=datetime.timedelta(seconds=5.0)):
to===>
 def fetch_into(self, num_samples, wfm, timeout=datetime.timedelta(seconds=5.0)):

closed 2018-03-13 by texasaggie97