ni/nimi-python #754: Fgen help

ihouses, 2018-03-01:
Hello,

I am trying the fgen module but I can not manage to take any waveform out. 
I am executing a simplified version of one example like:

********************
import nifgen
import time

def example(resource_name, options, waveform, frequency, amplitude, offset, phase, gen_time):

    with nifgen.Session(resource_name=resource_name, options=options) as session:

        session.output_mode = nifgen.OutputMode.FUNC

        session.configure_standard_waveform(waveform=nifgen.Waveform[waveform],        amplitude=amplitude, frequency=frequency,
                                            dc_offset=offset, start_phase=phase)

        with session.initiate():
            time.sleep(gen_time)

options = {'simulate': False, 'driver_setup': {'Model': '5433 (2CH)', 'BoardType': 'PXIe', }, }
example('FGEN', options, 'SINE', 1000, 1.0, 0.0, 0.0, 5.0)

********************

When I write simulate=True the script finishes but doesnt generate anything and when simulate is False the script crashes. I enclose the I/O trace

I have the same FGEN model than in the example PXIe 5433.

Thanks

[image]

marcoskirsch, 2018-03-01:
> When I write simulate=True the script finishes but doesnt generate anything

Sure, because you are running in simulation mode.

> and when simulate is False the script crashes. I enclose the I/O trace

It's actually that nifgen.Session.__init__() is raising, not that the script crashes, right? What is the text in the Exception?
Try calling this but don't pass any options, you shouldn't need that unless you are simulating.

ihouses, 2018-03-02:
Hi, well I have to say that even if simulate=True I was expecting to see some output, since I don't know what simulate exactly means and I expect that when you execute the given basic example works directly.

Somehow I managed to make work the example and now I see the output. But you know when we debug sometimes we dont close the session properly and the instrument remains iniatized.

Now I get the following error each time that I try to open a Session:
   raise Error(code, description)
nifgen.errors.Error: -1074134943: IVI:  (Hex 0xBFFA0061) The driver is already initialized.

I see that in NI MAX there is no option to reset the FGEN. What can I solve this problem without restarting the machine?

marcoskirsch, 2018-03-02:
> Hi, well I have to say that even if simulate=True I was expecting to see some output, since I don't know what simulate exactly means and I expect that when you execute the given basic example works directly.

You can read more about simulation [here](http://zone.ni.com/reference/en-XX/help/370524P-01/nifgenpropref/pnifgen_simulate/). But the short answer is that it allows you to write your programs with no hardware available. The program runs against a simulated device and you physical device is not touched or interacted with in any way.

> But you know when we debug sometimes we dont close the session properly and the instrument remains iniatized.

I don't know exactly what you are doing, but I'm very surprised you are running into such a problem. For all NI modular instrument APIs, if you open session b when session a is already open, the driver runtime closes session a and gives you session b. Furthermore, when the process exits the session is closed. You may have run into a driver bug. Since it's a different topic than the original one for this issue, I will ask the following from you:

Open a separate issue and include:

* NI-FGEN runtime version
* nifgen Python module version
* Python version and bitness
* Python script that reproduces the problem
* Device you are using
* OS version and bitness

Thanks

closed 2018-03-02 by marcoskirsch

marcoskirsch, 2018-03-07:
@ihouses you haven't opened an issue about `nifgen.errors.Error: -1074134943: IVI: (Hex 0xBFFA0061) The driver is already initialized.`
Is that because you can no longer reproduce the problem?