ni/nimi-python #795: Remove `signal_identifier` parameter from nidcpower `export_signal()`

texasaggie97, 2018-04-03:


texasaggie97, 2018-04-03:
This isn't as easy as I thought. The code generator does not currently support removing input parameters from the Python API and it will be complicated to add it because we still need the parameters for the call to the DLL.

Would removing them from documentation and making sure there is a default value be enough?

marcoskirsch, 2018-04-04:
This is true for niscope as well, right?
An alternative is to create "fancy" wrappers. I don't think it's worth it to be honest.

marcoskirsch, 2018-04-05:
> could remove the parameter from documentation and just let the default value handle it all the time.

I'd rather have documentation saying "reserved for future use" which is silly for a Python API but less confusing than not documenting it at all.

> The code generator does not currently support removing input parameters from the Python API and it will be complicated to add it because we still need the parameters for the call to the DLL.

Maybe more reasons to tackle this will emerge. If that happens, we can revisit this. I'm going to close.

closed 2018-04-05 by marcoskirsch